ZoneMinder/zoneminder #3963: Use only used material icons

connortechnology, 2024-04-11:
We include the entire set of material icons and font awesome icons, which is a fairly hefty download over a slow connection.

It is possible to generate a file with just the icons in use. So let's do that.

connortechnology, 2024-04-11:
@IgorA100 this would be a great job for you

IgorA100, 2024-04-12:
I don't think this is a good idea because... When adding a new icon to a page, will have to re-create new files each time.
But material-icons are really big.
I will think.

connortechnology, 2024-04-12:
We should be able to automate it.  If we just maintain a list.

IgorA100, 2024-05-14:
Not a bad idea, but I don't think it's the best option.
[Reduce Material Design Icons Font to 7KB and automate with PyAutoGUI](https://www.shedloadofcode.com/blog/reduce-material-design-icons-font-to-7kb-and-automate-with-pyautogui)