ZoneMinder/zoneminder #3966: Server shutdown when closing event in continuous recording mode

francescocarzaniga, 2024-04-14:
**Describe Your Environment**
- Zoneminder v1.37.59
- Installed via the master repo https://zmrepo.zoneminder.com/debian/master/
- Debian 12

**Describe the bug**
The camera is set up in continuous recording mode, camera passthrough, no analysis. Each section is recorded correctly but there is a significant amount of time (~20 seconds) between each section where nothing is recorded. When checking the logs it seems that Zoneminder is shutting down at the end of the section and then starting back up again, which causes ~20 seconds of delay.

**Debug Logs**
```
4/14/24, 7:22:47 PM GMT+2	zmc_m3	18508	INF	Opened new event 413201 Continuous	zm_monitor.cpp	2526
4/14/24, 7:22:47 PM GMT+2	zmc_m3	18508	INF	[Camera]: 002 - Opening event timestamp 1713115367 % 167	zm_monitor.cpp	2519
4/14/24, 7:22:44 PM GMT+2	zmstats	18491	INF	Stats Daemon starting in 30 seconds	zmstats.pl	-
4/14/24, 7:22:44 PM GMT+2	zmdc	18491	INF	'zmstats.pl' started at 24/04/14 19:22:44	zmdc.pl	-
4/14/24, 7:22:44 PM GMT+2	zmdc	18335	INF	'zmstats.pl' starting at 24/04/14 19:22:44, pid = 18491	zmdc.pl	-
4/14/24, 7:22:44 PM GMT+2	zmupdate	18480	INF	Update agent starting at 24/04/14 19:22:44	zmupdate.pl	-
4/14/24, 7:22:44 PM GMT+2	zmdc	18480	INF	'zmupdate.pl -c' started at 24/04/14 19:22:44	zmdc.pl	-
4/14/24, 7:22:44 PM GMT+2	zmdc	18335	INF	'zmupdate.pl -c' starting at 24/04/14 19:22:44, pid = 18480	zmdc.pl	-
4/14/24, 7:22:43 PM GMT+2	zmdc	18468	INF	'zmwatch.pl' started at 24/04/14 19:22:43	zmdc.pl	-
4/14/24, 7:22:43 PM GMT+2	zmdc	18335	INF	'zmwatch.pl' starting at 24/04/14 19:22:43, pid = 18468	zmdc.pl	-
4/14/24, 7:22:43 PM GMT+2	zmfilter_5	18443	INF	Scanning for events using filter id '5'	zmfilter.pl	-
4/14/24, 7:22:43 PM GMT+2	zmdc	18454	INF	'zmaudit.pl -c' started at 24/04/14 19:22:43	zmdc.pl	-
4/14/24, 7:22:43 PM GMT+2	zmdc	18335	INF	'zmaudit.pl -c' starting at 24/04/14 19:22:43, pid = 18454	zmdc.pl	-
4/14/24, 7:22:43 PM GMT+2	zmfilter_4	18430	INF	Scanning for events using filter id '4'	zmfilter.pl	-
4/14/24, 7:22:43 PM GMT+2	zmdc	18443	INF	'zmfilter.pl --filter_id=5 --daemon' started at 24/04/14 19:22:43	zmdc.pl	-
4/14/24, 7:22:43 PM GMT+2	zmdc	18335	INF	'zmfilter.pl --filter_id=5 --daemon' starting at 24/04/14 19:22:43, pid = 18443	zmdc.pl	-
4/14/24, 7:22:42 PM GMT+2	zmfilter_3	18418	INF	Scanning for events using filter id '3'	zmfilter.pl	-
4/14/24, 7:22:42 PM GMT+2	zmdc	18430	INF	'zmfilter.pl --filter_id=4 --daemon' started at 24/04/14 19:22:42	zmdc.pl	-
4/14/24, 7:22:42 PM GMT+2	zmdc	18335	INF	'zmfilter.pl --filter_id=4 --daemon' starting at 24/04/14 19:22:42, pid = 18430	zmdc.pl	-
4/14/24, 7:22:42 PM GMT+2	zmfilter_2	18405	INF	Scanning for events using filter id '2'	zmfilter.pl	-
4/14/24, 7:22:42 PM GMT+2	zmdc	18418	INF	'zmfilter.pl --filter_id=3 --daemon' started at 24/04/14 19:22:42	zmdc.pl	-
4/14/24, 7:22:42 PM GMT+2	zmdc	18335	INF	'zmfilter.pl --filter_id=3 --daemon' starting at 24/04/14 19:22:42, pid = 18418	zmdc.pl	-
4/14/24, 7:22:42 PM GMT+2	zmfilter_1	18390	INF	Scanning for events using filter id '1'	zmfilter.pl	-
4/14/24, 7:22:42 PM GMT+2	zmdc	18405	INF	'zmfilter.pl --filter_id=2 --daemon' started at 24/04/14 19:22:42	zmdc.pl	-
4/14/24, 7:22:42 PM GMT+2	zmdc	18335	INF	'zmfilter.pl --filter_id=2 --daemon' starting at 24/04/14 19:22:42, pid = 18405	zmdc.pl	-
4/14/24, 7:22:41 PM GMT+2	zmdc	18390	INF	'zmfilter.pl --filter_id=1 --daemon' started at 24/04/14 19:22:41	zmdc.pl	-
4/14/24, 7:22:41 PM GMT+2	zmdc	18335	INF	'zmfilter.pl --filter_id=1 --daemon' starting at 24/04/14 19:22:41, pid = 18390	zmdc.pl	-
4/14/24, 7:22:41 PM GMT+2	zmc_m3	18378	INF	Starting Capture version 1.37.59	zmc.cpp	221
4/14/24, 7:22:41 PM GMT+2	zmdc	18378	INF	'zmc -m 3' started at 24/04/14 19:22:41	zmdc.pl	-
4/14/24, 7:22:41 PM GMT+2	zmdc	18335	INF	'zmc -m 3' starting at 24/04/14 19:22:41, pid = 18378	zmdc.pl	-
4/14/24, 7:22:40 PM GMT+2	zmpkg	18302	INF	Single server configuration detected. Starting up services.	zmpkg.pl	-
4/14/24, 7:22:40 PM GMT+2	zmdc	18335	INF	Socket should be open at /run/zm/zmdc.sock	zmdc.pl	-
4/14/24, 7:22:39 PM GMT+2	zmdc	18335	INF	Server starting at 24/04/14 19:22:39	zmdc.pl	-
4/14/24, 7:22:38 PM GMT+2	zmpkg	18302	INF	Command: start	zmpkg.pl	-
4/14/24, 7:22:38 PM GMT+2	zmpkg	18302	INF	Sanity checking States table...	zmpkg.pl	-
4/14/24, 7:22:28 PM GMT+2	zmdc	15189	INF	Server shutdown at 24/04/14 19:22:28	zmdc.pl	-
4/14/24, 7:22:28 PM GMT+2	zmdc	15189	INF	'zmfilter.pl --filter_id=1 --daemon' exited normally	zmdc.pl	-
4/14/24, 7:22:28 PM GMT+2	zmdc	15189	INF	'zmfilter.pl --filter_id=2 --daemon' exited normally	zmdc.pl	-
4/14/24, 7:22:28 PM GMT+2	zmdc	15189	INF	'zmfilter.pl --filter_id=3 --daemon' exited normally	zmdc.pl	-
4/14/24, 7:22:28 PM GMT+2	zmdc	15189	INF	'zmupdate.pl -c' exited, signal 14	zmdc.pl	-
4/14/24, 7:22:28 PM GMT+2	zmdc	15189	INF	'zmfilter.pl --filter_id=4 --daemon' exited normally	zmdc.pl	-
4/14/24, 7:22:28 PM GMT+2	zmdc	15189	INF	'zmc -m 3' exited normally	zmdc.pl	-
4/14/24, 7:22:28 PM GMT+2	zmdc	15189	INF	'zmwatch.pl' exited normally	zmdc.pl	-
4/14/24, 7:22:28 PM GMT+2	zmdc	15189	INF	'zmaudit.pl -c' exited normally	zmdc.pl	-
4/14/24, 7:22:28 PM GMT+2	zmdc	15189	INF	'zmfilter.pl --filter_id=5 --daemon' exited normally	zmdc.pl	-
4/14/24, 7:22:28 PM GMT+2	zmdc	15189	INF	'zmstats.pl' exited normally	zmdc.pl	-
4/14/24, 7:22:27 PM GMT+2	zmdc	15189	INF	'zmwatch.pl' sending stop to pid 15326 at 24/04/14 19:22:27	zmdc.pl	-
4/14/24, 7:22:27 PM GMT+2	zmdc	15189	INF	'zmupdate.pl -c' sending stop to pid 15338 at 24/04/14 19:22:27	zmdc.pl	-
4/14/24, 7:22:27 PM GMT+2	zmdc	15189	INF	'zmc -m 3' sending stop to pid 15232 at 24/04/14 19:22:27	zmdc.pl	-
4/14/24, 7:22:27 PM GMT+2	zmdc	15189	INF	'zmfilter.pl --filter_id=4 --daemon' sending stop to pid 15286 at 24/04/14 19:22:27	zmdc.pl	-
4/14/24, 7:22:27 PM GMT+2	zmdc	15189	INF	'zmfilter.pl --filter_id=3 --daemon' sending stop to pid 15271 at 24/04/14 19:22:27	zmdc.pl	-
4/14/24, 7:22:27 PM GMT+2	zmdc	15189	INF	'zmfilter.pl --filter_id=1 --daemon' sending stop to pid 15243 at 24/04/14 19:22:27	zmdc.pl	-
4/14/24, 7:22:27 PM GMT+2	zmdc	15189	INF	'zmfilter.pl --filter_id=2 --daemon' sending stop to pid 15257 at 24/04/14 19:22:27	zmdc.pl	-
4/14/24, 7:22:27 PM GMT+2	zmdc	15189	INF	'zmstats.pl' sending stop to pid 15348 at 24/04/14 19:22:27	zmdc.pl	-
4/14/24, 7:22:27 PM GMT+2	zmdc	15189	INF	'zmfilter.pl --filter_id=5 --daemon' sending stop to pid 15298 at 24/04/14 19:22:27	zmdc.pl	-
4/14/24, 7:22:27 PM GMT+2	zmdc	15189	INF	'zmaudit.pl -c' sending stop to pid 15310 at 24/04/14 19:22:27	zmdc.pl	-
4/14/24, 7:22:27 PM GMT+2	zmstats	15348	INF	Stats Daemon exiting	zmstats.pl	-
4/14/24, 7:22:27 PM GMT+2	zmstats	15348	INF	Received TERM, exiting	zmstats.pl	-
4/14/24, 7:22:27 PM GMT+2	zmfilter_1	15243	INF	Received TERM, exiting	zmfilter.pl	-
4/14/24, 7:22:27 PM GMT+2	zmdc	15189	INF	Server exiting at 24/04/14 19:22:27	zmdc.pl	-
4/14/24, 7:22:27 PM GMT+2	zmaudit	15310	INF	Received TERM, exiting	zmaudit.pl	-
4/14/24, 7:22:27 PM GMT+2	zmfilter_3	15271	INF	Received TERM, exiting	zmfilter.pl	-
4/14/24, 7:22:27 PM GMT+2	zmfilter_5	15298	INF	Received TERM, exiting	zmfilter.pl	-
4/14/24, 7:22:27 PM GMT+2	zmc_m3	15232	INF	[Camera]: image_count:5922 - Closing event 413200, shutting down	zm_monitor.cpp	3573
```

francescocarzaniga, 2024-04-21:
@connortechnology This problem is due to the systemd service. It cannot detect that the process has forked due to the following error:
``` New main PID does not belong to service, and PID file is not owned by root. Refusing. ```
So it restarts every 600 seconds as indicated in `Timeout`.

connortechnology, 2024-04-21:
I run several debian 12 servers here... with no issues like this.
Have you made any modifications to the .service file? Please post here your /etc/systemd/system/zoneminder.service

francescocarzaniga, 2024-04-21:
No modifications at all, this is a fresh install using the `https://zmrepo.zoneminder.com/debian/master` repo. 
Contents of `/lib/systemd/system/zoneminder.service`:
```
# ZoneMinder systemd unit file
# This file is intended to work with Debian distributions

[Unit]
Description=ZoneMinder CCTV recording and surveillance system
After=network.target mysql.service
# Remarked out so that it will start ZM on machines that don't have mysql installed
# Override it by placing an override.conf in /etc/systemd/system/zoneminder.service.d
#BindsTo=mysql.service

[Service]
#User=www-data
Type=forking
ExecStart=/usr/bin/zmpkg.pl start
ExecReload=/usr/bin/zmpkg.pl restart
ExecStop=/usr/bin/zmpkg.pl stop
PIDFile=/run/zm/zm.pid
Restart=always
RestartSec=10
Environment=TZ=:/etc/localtime
TimeoutSec=600

[Install]
WantedBy=multi-user.target
```
Now I have modified it to be a oneshot just to make it work, but it's a temporary fix.