ZoneMinder/zoneminder #3968: Timestamp is not shown in the video picture

schnudd31do3, 2024-04-17:
- Zoneminder 1.37.59
- Installed via deb package (zoneminder_1.37.59~20240417.30-bookworm_amd64.deb)
- Debian bookworm 12.5
- Edge, Chrome, Firefox

- Camera Axis M3005

The time stamp configured here:
[image]

is not shown in the video picture. 
[image]

In the official debian version 1.36.33, it is functioning.

ovargasp, 2024-04-21:
schnudd31do3, thanx for the report, I was able to reproduce the behavior, there is in fact a problem reported in the logs about that, it states that a font is missing... that's why you're not seeing the time stamp.

[image]

A new release will fix the problem, but meanwhile there is a work around you can apply: download the file and place it in the corresponding forder. 

The file is located here: https://github.com/ZoneMinder/zoneminder/blob/master/fonts/default.zmfnt

Download it, move it to the corresponding folder and restart zoneminder service so it gets recognized, that'll fix it.

connortechnology, 2024-04-21:
Why was it missing I wonder?  Should have been installed automatically.

schnudd31do3, 2024-04-22:
I tried it twice. I restored my server to the state before updating and it was always the same result.
I will test the work around from https://github.com/ZoneMinder/zoneminder/blob/master/fonts/default.zmfnt

schnudd31do3, 2024-04-22:
There is one question left. Why does the camera image only appear with a delay of about a minute after the URL is opened in the browser? https://mycamserver.de/zm/cgi-bin/zms?mode=jpeg&monitor=1&user=MyUser&pass=MyPass

ovargasp, 2024-04-22:
schnudd31do3, I'm glad you got your first issue fixed, about this second question, to try and figure out the problem is necessary to have some more information, logging, what browsers have you tried, etc. 

I tried to replicate what you're saying but it's working fine in my Debian installs...

ovargasp, 2024-04-22:
Can you open the monitor view and right click on top of  the stream then select "copy the image URL", that will give you a slightly different URL that the one you're using, try and use that one in another browser tab... it's important to leave UI monitor view prior to try and open this URL, because it will use the same randomizer number that the monitor view and it will delay the playback.

connortechnology, 2024-04-22:
I think we would need to turn on debug logging and check the zms_m1.log

schnudd31do3, 2024-04-23:
Thanks. I got you. In my case it is : https://mycamserver.de/zm/cgi-bin/nph-zms?mode=jpeg&maxfps=&scale=100&state=1&monitor=1&auth=1315899ba107d0c2d0625ae643be8b5e&user=admin&rand=1713846240&connkey=308082. Opened in a separate tab does not show the video picture.

schnudd31do3, 2024-04-23:
No errors, only waiting for responding.
[image]

schnudd31do3, 2024-04-23:
I observed, restarting the zoneminder service causes about 4 seconds of delay while the website is kep open. Restarting the apache2 service causes a delay of about 36 seconds, where I have to refresh the website immediately after restarting the apache2 service.

ovargasp, 2024-04-24:
Have you tried to access the monitor using the browser in "safe mode", that may give you an idea if is related to some browser extesion... another thing you can check is if the CPU usage spikes when you open the monitor view, that can cause a delay.

schnudd31do3, 2024-04-25:
1. I dont know how to activate "safe mode". I am not running it in a docker container.
2. When opening with "mode=single", the picture ist shown immediately and by presing F5 as fast as I want, the video image ist updated without any delay. But using "mode=jpeg", the delay is there. Because cgi-bin/zms is a compiled file, I am not able to dig in, what the delay causes. I guess, that zms is buffering some frames before it shows the video image. An indication for that is that when stopping the zoneminder service, the video image continues for a few seconds.
3. I am using 4 core with 2 CPU per socket and 4GB RAM. vSphere shows no spike when opening the URL and is running on about 15% of CPU load. So I dont beleve that there is a lack of power.

Can you confirm 2. from above? If so, where can I disable this behaviour? I know for a fact that this behavior wasn't the case until some previous versions and I did not change my hardware

connortechnology, 2024-04-25:
The answers will be in the zms logs. The CSP stuff is caused by some chrome extension. I don't think it is related.  However you may try turning off extensions one by one until that goes away.

A bigger issue is a misconfig on your logging path.. it can't write to /var/log/zoneminder.log.  This is a strange value.  Logs should be in /var/log/zm/ so something is strange with your config.  Please check /etc/zm/conf.d/01-system-paths.conf

connortechnology, 2024-04-25:
zms does not buffer frames.  It waits for zmc to have captured another image, then sounds that out.  There is a weird thing with Chrome where it doesn't display an image until it gets a second one.  For this reason we actually sometimes send the same image twice.  However I don't think this is relevant.

schnudd31do3, 2024-04-30:
I checked this 01-system-paths.conf. I had the path configured here: ZM_PATH_LOGS=/var/log/zm. Perhaps it was an old error.

You said "It waits for zmc to have captured another image". But why can I capture immediately images as fast as I want by using the single mode and pressing F5? Where can check, on what zns is waiting for when using the jped mode?

connortechnology, 2024-04-30:
I just saw this problem today on a support call.  Despite my best efforts to figure it out, so far I am stumped.  They had two identical servers and one did it while the other did not.  Independent of browser.  zms logs clearly show images being sent, but the browser refuses to display them for around a minute.  Also seems to not receive sigpipe when we browse away.  Very puzzling.  Logs all green.

schnudd31do3, 2024-04-30:
Thanks, that you could confirm my observations. "ogs all green." yes they are. I am pretty sure that in former versions of zms, it does not have this issue. But I could not remember since when it comes up.

SteveGilvarry, 2024-09-13:
Closing out as original issue with missing font was resolved. Please create a new issue to track browser delays secondary issue you experienced or if @connortechnology wants one created to track that issue

closed 2024-09-13 by SteveGilvarry

connortechnology, 2024-09-13:
Just wanted to add an update RE my support call.  That issue was resolved after discovering that the second server didn't have as many cpu cores.  (Dual socket, only 1 socket filled).  Adding another cpu solved that problem.

Doesn't really explain much though.  mode=single uses a different code path and just sends out whatever the last image was and exits. mode=jpeg sends the last image and then waits for the next.  Also, there is the communication socket setup so that we can send STOP/PAUSE get fps rates etc.