ZoneMinder/zoneminder #3972: High CPU Usage causing reboot with 126 IPCs

zzyyzz1992, 2024-04-22:
**Describe Your Environment**
- Version of ZoneMinder [release version]
- How you installed ZoneMinder [PPA]
- Full name and version of OS Ubuntu 2204(Freshly installed)

**Describe the bug**
I have 126 H264 onvif ip cameras previesly handled by a hardware NVR, now I'm trying to migrate all the cameras to zoneminder, when using record mode without 'Decoding Enabled', CPU workload is below 10, Mem usage is around 12GB, networking(RX) is about 24MiB. 
Everything was fine.

But when using record mode with 'Decoding Enabled', the zoneminder quickly consumes all my CPU and Mem, then the Swap(2GB) , all happens in just dozens of sencond, after that it seems the zoneminder got restarted, and again consuming, restarting ....
Cuz my zm is in a vm, so I enlarged the Mem to 256GB(My physical machine has dual amd 7601 32cores and 512GB Mem), but the zm just quickly consumes all resouces and couldn't avoid timelessly restarting.

I just wondering that the cause is my configure wrong, or something else? Dose a discrete GPU help(I got a dust-eating GTX1060)?

I'd appreciate your help!

connortechnology, 2024-04-22:
126 cams is a lot. You don't mention resolution, but at 720p we generally assume 4 cams per core if doing motion detection.

Do you need decoding support?  You can install rtsp2web or janus and use direct from camera viewing, removing the need  for decoding.  Also there is the keyframes only or ondemand decoding.  

You might want to consider purchasing a few hours of support from us to help get this tuned in.

SteveGilvarry, 2024-09-13:
Closing as no response from OP

closed 2024-09-13 by SteveGilvarry