ZoneMinder/zoneminder #3973: Camera FPS in Montage and Montage review are not the same

Saske123GG, 2024-04-26:
(Hi, guys. Rip English, sorry for that >;)
Different FPS on cameras between Montage and Montage Review

- Version of ZoneMinder v1.37.59
- I installed the program according to the instructions from the official YouTube ZM channel
- Ubuntu Server 22.04 LTS
- Google Chrome browser

Cameras:
- Hikvision
- 22 FPS on each
- The first camera has a resolution of 1920x1080, the second camera has 1280x720, the third camera has 1520x1520.
- RTSP - FFMPEG - H264

My problem:
When viewing the cameras (via browser) in real time (Montage), they have a frame rate of about 22 frames per second.
But when I look at the events in the "Montage Review" tab, the events are played back at 1fps. Why?

Is this how it should work, or can it be set to have events in Montage Review play back at 22 frames per second, just like everywhere else?

I ask because ZmNinja's Montage Review mode plays these events at normal frame rates.

connortechnology, 2024-04-26:
montage review in zm is .... interestingly designed.  It was designed to work around the browser connection limit, so instead of streaming jpegs it is asking for individual jpegs.  This creates a lot of overhead.  It has to keep them all in sync so it keeps a timer and constantly asks for the image at that time.  It needs work.  I have been improving it for years, but it still has a long way to go.

momingweixiao, 2025-03-04:
Hello, I have also encountered the same problem. Is there any method or setting to solve the abnormal frame rate of "Montage Review" playback? Currently, the display shows: Display refresh rate is 0.7 per second, avgFrac=0.420. The screen looks very laggy.

SteveGilvarry, 2026-01-02:
Closing as explained by design in the current Montage Review, and will likely need a completely new approach to address issues.

TheBlueMatt, 2026-01-02:
Is there an issue to track taking a new approach that users can subscribe to?

SteveGilvarry, 2026-01-02:
@TheBlueMatt Unfortunately, the project does not have a feature tracker, and with only one developer (Icon) doing regular development work, the number of stale issues has gotten out of control. I'm helping out while on holiday by closing issues that are fixed already, need more details or unlikely to be fixed and letting people know the status. 

This won't be addressed in 1.38 (the next release), so version 1.40 would be the earliest this might appear, when we build a streaming solution that is not impacted by browser limitations and very old technology.

Any future developments on this will be announced first in Slack/Discord and then in the release notes. Reach out on Slack or Discord if you want to discuss more or can help.