ZoneMinder/zoneminder #3974: Filter does not correctly calculate the available space in the storage

IgorA100, 2024-04-27:
**Describe Your Environment**
ZoneMinder 1.37.58

Storage is connected as a shared share in Ubuntu

[image]
[image]
[image]

Filter selection includes all events from the storage.

connortechnology, 2024-04-28:
So... NAS is at 80% and your filter is set to list events when it is above 77%.  And it does.  What am I missing?

IgorA100, 2024-04-28:
> What am I missing?

The filter selected ALL events available on the NAS. This can be seen by date (sorting column) and quantity (bottom left)

connortechnology, 2024-04-28:
So the point is that it isn't honouring the 100 limit. Nothing to do with disk percent.

IgorA100, 2024-04-28:
Oh dear Isaac, I've overworked, I'll go and rest.
I'm sorry.

closed 2024-04-28 by IgorA100