ZoneMinder/zoneminder #3983: Support FFMPEG 7

VVD, 2024-05-06:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit] 1.36.33
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]: from FreeBSD ports
- Full name and version of OS: FreeBSD 14.0, 13.3 amd64.
- Browser name and version (if this is an issue with the web interface): -

**If the issue concerns a camera**
No.

**Describe the bug**
Build failed with FFMPEG 7.
All deprecated API in FFMPEG 5/6 was removed from FFMPEG 7.

**To Reproduce**
Steps to reproduce the behavior:
1. Try to build with FFMPEG 7.

**Expected behavior**
Build and work without errors.

**Debug Logs**
```
In file included from /wrkdirs/usr/ports/multimedia/zoneminder/work/zoneminder-1.36.33/src/zm_monitor.h:24:
/wrkdirs/usr/ports/multimedia/zoneminder/work/zoneminder-1.36.33/src/zm_camera.h:111:51: error: no member named 'channels' in 'AVCodecParameters'
    return mAudioStream ? mAudioStream->codecpar->channels : -1;
                          ~~~~~~~~~~~~~~~~~~~~~~  ^
1 error generated.
```
Full build log: https://pkg-status.freebsd.org/beefy22/data/140amd64-default-foo/2024-05-04_07h10m46s/logs/errors/zoneminder-1.36.33_3.log

VVD, 2024-05-06:
This can help to update: https://code.videolan.org/videolan/vlc/-/merge_requests/5304/diffs
Maybe replace `mAudioStream->codecpar->channels` with `mAudioStream->codecpar->ch_layout.nb_channels`.

VVD, 2024-05-06:
```
/wrkdirs/usr/ports/multimedia/zoneminder/work/zoneminder-1.36.33/src/zm_mpeg.cpp:632:38: error: no member named 'frame_number' in 'AVCodecContext'; did you mean 'frame_num'?
  632 |                 opicture_ptr->pts = codec_context->frame_number;
      |                                                    ^~~~~~~~~~~~
      |                                                    frame_num
/usr/local/include/libavcodec/avcodec.h:2030:13: note: 'frame_num' declared here
 2030 |     int64_t frame_num;
      |             ^
2 warnings and 1 error generated.
*
```
https://git.ffmpeg.org/gitweb/ffmpeg.git/commitdiff/6b6f7db81932f94876ff4bcfd2da0582b8ab897e
https://git.ffmpeg.org/gitweb/ffmpeg.git/commitdiff/2717dcfb8527dd65a9fb0e7f8cb00d118af65b68

connortechnology, 2024-05-08:
I have updated both master and release-1.36 with further ffmpeg7 deprecation handling.

connortechnology, 2024-05-14:
Hey I switched to using quick_exit instead() of exit(). this breaks on openbsd for me.  Just wondering if it has broken freebsd as well.  Does ZM still c ompile for you?

VVD, 2024-05-15:
Do u say about this commit: https://github.com/ZoneMinder/zoneminder/commit/0048a6bc9972abc5a76a659d1cfd7e0423b56f17?
I have only one host in production with zoneminder - can't test run development versions.
I can test build only.

connortechnology, 2024-05-15:
Yes that's the one.  quick_exit is not implemented in openbsd.  I'm testing a workaround of using abort() instead.  Just wondering if quick_exit exists in freebsd.

VVD, 2024-05-15:
Look like it work fine:
```
$ cat e.cpp
#include <cstdlib>
#include <iostream>

int main()
{
  std::cout << "Hello world!\n";
  std::quick_exit(1);
}
$ clang e.cpp -o e -lc++
$ ./e
Hello world!
$ echo $?
1
```

connortechnology, 2024-05-15:
Good to know, thanks for testing!

FYI on openBSD we get:
# clang e.cpp -o e -lc++
e.cpp:7:8: error: reference to unresolved using declaration
  std::quick_exit(1);
       ^
/usr/include/c++/v1/cstdlib:145:1: note: using declaration annotated with 'using_if_exists' here
using ::quick_exit _LIBCPP_USING_IF_EXISTS;
^
e.cpp:7:3: warning: expression result unused [-Wunused-value]
  std::quick_exit(1);
  ^               ~
1 warning and 1 error generated.

emaste, 2024-05-23:
`std::quick_exit` is C++11 / C11 so I'd expect the OpenBSD folks will want to implement it.

They could presumably take the FreeBSD implementation from https://github.com/freebsd/freebsd-src/commit/0a31efe016d28d5f3bee6c24591831149d760650

connortechnology, 2024-05-23:
They don't seem to want to.  I am using abort instead on openbsd. For now.

synthead, 2024-08-07:
It'd be great to prioritize this, if possible.  ffmpeg 7 is now in the mirrors for Arch Linux, so the zoneminder AUR package is currently broken:

- https://aur.archlinux.org/packages/zoneminder

It would be possible to put in the effort to create a custom ffmpeg 6 package that the zoneminder AUR package uses, or possibly vendor ffmpeg with zoneminder.  However, if ffmpeg 7 support is already in the master branch, then all we need is another release to build against stable zoneminder sources.

Thanks in advance!

connortechnology, 2024-11-27:
1.35.35 was released.  We can close this.