ZoneMinder/zoneminder #3989: Improve UX

olignyf, 2024-05-10:
**Describe the bug**
UX is due for a refresh and could be more intuitive.

**Expected behavior**
 - Configurations could be easier to find and/or new simplified operations/workflow could be added. 
 - Labels and wording could be improved at many place to better describe the option's behavior
 - More modern CSS look & feel.
 - Maybe throw in one or two graphs of disk usage and stats on alert, etc.

For example (but not limited to) these operations could be improved:
 - How to record in lower resolution than the incoming stream. At the moment we can change the "Source > Capture Resolution (pixels)" resolution which affect both preview resolution and record resolution. This could be split into two field, the "Capture Resolution" and the "Record Resolution" as a new distinct control. The default for "Capture Resolution" should be something like "Auto-Detect/Actual".
- The delete storage operation could have additional option to purge the file on disk as well.
- The storages could have new options to limit storage % usage.
- A "Test" button for the FTP & SFTP configuration.

Thanks!

IgorA100, 2024-05-11:
> Configurations could be easier to find and/or new simplified operations/workflow could be added.

It's not clear what exactly you don't like. Can you explain in more detail?

olignyf, 2024-05-13:
> It's not clear what exactly you don't like. Can you explain in more detail?

Check the "For example (but not limited to) these operations could be improved" part of the issue description for some example. I can improve the list if needed.

In general I'm certain a UX person could review the workflow and come up with many improvement proposals. Right now it looks like everything was just added in various places as a first try reflecting where it is in the code without much thought on usability. One could ask a few persons which area the have some struggle with and could be improved. If you ask me I would say configuring the **recording resolutions**, **managing events thresholds**, and **storage management**. I might create a separate ticket for the storage issue I came across. By default when I installed ZoneMinder it quickly filled up my HD. So I went to create a new storage config hoping I could set it to limit at 50% of my disk space, but there was no such option in Storage. I assigned my camera to the new storage. Then I figure out I needed to go in "Filter" the change the maximum and that it's a global setting. Again "Filter" is probably the programmatic term used in the code I presume and not the best clearest choice for a user interface. Why wouldn't "Options > Storage" and "Filter" not be in a similar top level "Storage" page? To go back to my story I erased my new storage config hoping it would take me to how it was by default, and made sure my camera storage was not configured to the erased one. It says it uses "Default". But now all my events are gone, and even if I setup motion detect and it appears to be writing new event, the list stays empty. I don't even know how to recover from this situation. I tried to create the storage again and assign it to the camera but it stays empty. Maybe I did something wrong and the interface could have prevented me to do so?

Also there are various simple UI improvements to do such as this one for example:
If I click on the name of the camera it takes to configuration. The link could be centered veristically with the white space above and below and with respects to the square buttons on the left. Also a gear icon would makes it clear what it does and that it's clickable. 
[image]
If I click it it takes me to the configuration. Then we could be offered a similar link at the same place to go back to monitoring. The back button functionality takes you back to the previous place which may or may not be the monitoring. So would be nicer to have a similar header link to go back to monitor with maybe a monitor icon. 
![zm link configure]
(https://github.com/ZoneMinder/zoneminder/assets/1154566/82ecd584-9fa7-41b8-add7-da9732ec6941)

Also when you are the Console, to edit a camera you need two steps: Check the checkbox, then click "Edit". A gear icon in the row would be faster.
[image]

IgorA100, 2024-05-14:
Thanks for your detailed summary.
ZM is constantly improving and version 1.37 has a lot of changes compared to 1.36.
Please try version 1.37 https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder-master

In order to open the monitor editing page, you need to click on the IP address. (There is a gear in the column header)

"Filter" & disk cleanup - yes, I agree, changes need to be made here.
@connortechnology
I suggest adding a more intuitive option in the storage settings.