ZoneMinder/zoneminder #3997: new montage/view changes don't update scale in url

connortechnology, 2024-05-15:
Hence we are streaming a full res image, then scaling it down in the browser.  This takes a ton of CPU to encode high res jpgs which aren't even really used.

When we scale the img tag, we need to update the src= and the scale= parameter.

IgorA100, 2024-05-15:
> Hence we are streaming a full res image, then scaling it down in the browser.

After the image is displayed in the browser, the "setScale" method is called, which calculates the required "scale" depending on the actual size of the image in the browser. As far as I understand, this works correctly, because... when the image is small in the browser, the "setScale" method generates a "scale" from 25 to 40, when the image is enlarged, the "scale" increases to 100. This can be checked visually like this:
We go to the Layout editing mode and make the image small, for example 200px in width, then stretch the image, for example, to 1000px in width and see that at first the image resolution is low, but after a moment the resolution increases.
As far as I understand, the "scale=" in "src" does not affect the "scale" generated in "setScale".
The "setStreamScale" method also ignores the "scale=" in "src" if there is a 'connkey' and 'mode=single' in the "src".
1. I tried changing "scale=" in "src" and it did not lead to any changes.
2. Before the introduction of the new PanZoom, "scale=" in "src" also did not change and was always = 100

These are my observations.
I may be wrong somewhere and any additional information is welcome.

IgorA100, 2024-05-15:
Maybe I'm confused about something, I'll double check
I tested resizing on Montage page
I'll check more thoroughly on the Watch page

PR #3998 (2024-05-15): Fix: "Scale" calculation algorithm (skin.js)
Close #3997
closed 2024-05-15 by connortechnology