ZoneMinder/zoneminder #4005: When viewing a recorded event, the stream stops when viewing ends and does not resume when you click "Play"

IgorA100, 2024-05-16:
**Describe Your Environment**
- Version of ZoneMinder 1.37

**To Reproduce**
Steps to reproduce the behavior:
1. Enable viewing of a recorded event lasting 1 minute
2. Go have a coffee, come back in 5-10 minutes
3. Press the "Play" button to watch again.
4. Viewing is not available, there is an error in the browser console

> js_logger-base-1712621798.js:91 getCmdResponse stream error: socket_sendto( /run/zm/zms-002076s.sock ) failed: Connection refused

**Expected behavior**
After pressing the "Play" button, playback should begin.

connortechnology, 2024-05-16:
the error message means that zms exited.  Maybe check in zms logs for why. Because it should hang around on pause.  However I can simulate the situation by manually killing zms.

closed 2024-05-16 by connortechnology

IgorA100, 2024-05-16:
Pause works - OK
zms exited after the event has finished playing - OK
But after clicking on the "Play" button, you need to restart the zms process.

commit af233ef9 (2024-05-16): Restart stream when playClicked and zmsBroke. Fixes #4005