ZoneMinder/zoneminder #4006: Video.js используется устаревший 4.11.4, на данный момент есть версия 8.12.0

IgorA100, 2024-05-17:
**Describe Your Environment**
- Version of ZoneMinder 1.37

http://videojs.com
https://github.com/videojs/video.js

IgorA100, 2024-05-17:
@connortechnology 
I'll see when I have time

connortechnology, 2024-05-17:
For consistency of UI, video.js may not really be the way to go.  Might be better to integrate something like https://github.com/numberwolf/h265web.js which would give us h265 as well.

IgorA100, 2024-05-17:
Agree.
I will think.

connortechnology, 2024-05-17:
See some partial work I did here https://github.com/connortechnology/ZoneMinder/tree/h265web.js

IgorA100, 2024-05-17:
> partial work

Why didn't you finish your work? What were the difficulties?

connortechnology, 2024-05-17:
I think it basically worked, just needed to hook up all the UI elements