ZoneMinder/zoneminder #4036: Feature Todo list: Montage Review

connortechnology, 2024-05-23:
Montage Review is fairly broken in it's design.  It currently constantly grab images one at a time and writes them to a canvas.  It seems that this cannot handle an mjpeg stream.  

The reason for the canvas is to be able to write text and the +/- buttons onto it.  I think we should replace it with the new layout from montage.  

OR

We copy over the history timeline logic to montage and scrap review.

IgorA100, 2024-05-23:
>  canvas is  to be able to write text

I don’t quite understand what kind of text you are writing about?

connortechnology, 2024-05-23:
If it can't load event data it writes "No data"over the image.  Really not needed. Can be done other ways.

IgorA100, 2024-05-23:
Ok.

IgorA100, 2024-05-23:
Is Live mode required? 
In essence, this is a duplication of the Montage page, but with FPS = 1.
FPS = 1-5 - this would be very good for a narrow network channel on both Montage and Montage Review

connortechnology, 2024-05-23:
No it is not.  I would really like to merge these two views.  At the moment montage is much better.

I will be committing a branch tomorrow that adds an fps limiter to montage.

IgorA100, 2024-05-25:
Why not make a single page for live viewing and for viewing recordings?
For example, you can make such a page based on Montage page.

IgorA100, 2024-06-19:
@connortechnology 
I can't figure out where in PHP the "ZM_OPT_CONTROL" constant is defined?
I can't access it when working via AJAX.

IgorA100, 2024-06-19:
@connortechnology 
What is this button for and does it work?
[image]

When you press it, the camera control panel appears, but there is no image from the camera.
[image]

connortechnology, 2024-06-19:
I think maybe you deleted the question about ZM_OPT_CONTROL.  For js land, all is provided by web/skins/classic/js/skin.js.php.  All but a few config variables are exposed there.

For php it is in web/includes/config.php.in (which becomes config.php).

IgorA100, 2024-06-19:
> It is a leftover from the past.
...... perhaps it could just go away.

Got it, I’ll leave it “as is” for now, I won’t delete it. I thought that maybe something was broken on me. I also don’t see the point in this page if the camera can be controlled on the Watch page.

connortechnology, 2024-06-19:
The only point would be to do so while watching multiple feeds.  Which is a valid use case, maybe you have multiple cameras from different angles watch a scene...

IgorA100, 2024-06-19:
> The only point

Yes, it’s possible, but now I definitely won’t do it, because... a lot of work on partially converting the Montage page to Ajax and integrating it with Montage Review.

IgorA100, 2024-06-19:
> I think maybe you deleted the question about ZM_OPT_CONTROL.

Yes, I deleted the message because... I have access to the ZM_OPT_CONTROL constant in my PHP code, but I still don’t understand where it is defined.

> For php it is in web/includes/config.php.in (which becomes config.php).

The fact of the matter is that there is no ZM_OPT_CONTROL constant in config.php
It probably comes from SQL somewhere and must be defined in PHP somewhere.

connortechnology, 2024-06-19:
It is in the database.
config.php loads /etc/zm.conf to find out where the database is. Then loads the contents of the Config table.