ZoneMinder/zoneminder #4037: The placement of monitors on the Montage page is broken again

IgorA100, 2024-05-23:
**Describe Your Environment**
- Version of ZoneMinder 1.37

Someone somewhere broke something...... #4032

connortechnology, 2024-05-23:
What is happening is that my 1080p cameras end up with a width of 433px. My 640x480 cameras end up with a width of 430.  So the ratio calculation results in a different height value. 

It is not clear to me how the width could be different. Perhaps that points you in the right direction.

IgorA100, 2024-05-23:
Are the heights of these cameras the same or different?

IgorA100, 2024-05-23:
Check it like this:
1. layout = "4Wide" then refresh the page (Ctrl+F5)
2. Switch Layout="3Wide"
3. Refresh page (Ctrl+F5) (The previous selection 3Wide should remain)
4. Switch Layout="4Wide"

How will the cameras be positioned after each of the 4 actions?

connortechnology, 2024-05-23:
THe source heights are different 1920x1080 vs 640x480.  The resulting heights after scaling are different because the widths somehow end up different. 275px vs 279px.

connortechnology, 2024-05-23:
They are also a mix of img and video tags. May be related.

connortechnology, 2024-05-23:
3 Wide lays out correct immediately and after refresh.  4 out lays out correctly after change to 4 wide but incorrectly on refresh.  Again, it may be an interaction with the video tag.

PR #4039 (2024-05-24): Fix:  changedMonitors.forEach (montage.js)
Iterating through the array was incorrect

Possible fixes #4037
closed 2024-05-24 by connortechnology