ZoneMinder/zoneminder #4041: We need to not even try to load cameras that are off view.

IgorA100, 2024-05-27:
Indeed, an interesting idea. I propose to discuss.
But at the moment I can't stop the camera on the Montage page.
Neither does `monitors[i].streamCommand(CMD_QUIT)`
Neither does `monitors[i].stop()`
Neither does `monitors[i].kill()`

IgorA100, 2024-05-27:
@connortechnology 
Think about how you can stop it, and then, if necessary, start the monitor again on the Montage page.
I'll try to implement the idea when only those monitors that are in view in the browser are launched. So as not to overload the server with work.

connortechnology, 2024-05-27:
I have a branch here where I started work on it. You may wish to reference it.

https://github.com/connortechnology/ZoneMinder/tree/only_stream_visible

IgorA100, 2024-05-27:
I'll see.
I didn't think you'd started this work yet.

IgorA100, 2024-05-28:
@connortechnology 
Your code is almost ready, I won't bother you.
A couple of points:
- For STICKY mode the following is required:
`document.querySelector('#content').onscroll = on_scroll;`
- Stopping monitors does not work for me.