ZoneMinder/zoneminder #4043: Rate=1 on Watch page does not work

IgorA100, 2024-05-27:
**Describe Your Environment**
- Version of ZoneMinder 1.37

The other Rate options work.

IgorA100, 2024-05-28:
> Works for me although I have seen some weird behaviour with the stream sometimes, not dependent on fps

Hm. Additional analysis is probably required.

> What doesn't work is Stills mode.

I didn’t even know what kind of regime this was, but now I understand. Perhaps it should be removed from the selection?

> Seems to me we may need rate options less than 1. Like update every 10 seconds or every minute.

Maybe. 
Just tried it. When the FPS value is less than 1 (as with =1), a black screen with the words "Waiting for capture" appears, and then a picture appears, then again a black screen and again a picture. Not pretty at all.

connortechnology, 2024-05-28:
Yeah I have seen that behaviour but cannot at this moment reproduce it.  Log analysis will be required.  Waiting for capture means that zmc isn't decoding.

IgorA100, 2024-05-28:
If "decode = All" - the black screen appears less frequently, if "decode = On demand + key" - the black screen appears more often. And it’s like this on all my cameras...
There are no errors in the logs, I did not enable debugging.

closed 2024-09-19 by IgorA100