ZoneMinder/zoneminder #4051: An error appears when switching from Montage page to Console page.

IgorA100, 2024-05-29:
**Describe Your Environment**
- Version of ZoneMinder 1.37

`ERR | Error writing packet: Invalid argument | zm_videostore.cpp`

At the moment the Montge page is closed, something probably needs to be stopped.

IgorA100, 2024-05-29:
In montage.js there is code in "window.onbeforeunload = function(e) {" that stops the monitors.
Why is this code delayed?
Were there any problems?

IgorA100, 2024-05-29:
Why not leave the function like this: ?
```
window.onbeforeunload = function(e) {
  console.log('unload');
  //event.preventDefault();
  for (let i = 0, length = monitorData.length; i < length; i++) {
    monitors[i].kill();
  }
};
```

IgorA100, 2024-05-29:
Или вот более правильный код:
```
document.onvisibilitychange = () => {
  if (document.visibilityState === "hidden") {
    for (let i = 0, length = monitorData.length; i < length; i++) {
      monitors[i].kill();
    }
  }
};
```

connortechnology, 2024-05-29:
It is code that I was trying out. When the browser is overloaded, trying to navigate away takes forever.  So this function is supposed to stop the streams so that the browser can actually get around to navigating away.

connortechnology, 2024-05-29:
I highly doubt that any message from videostore has anything to do with navigating away from the montage view.  Normally those messages are caused by changes in timestamps back in time, although our current code should not allow that.

PR #4053 (2024-05-29): Stop monitors when closing montage page
Possibly closed #4051