ZoneMinder/zoneminder #4070: Thoughts on event view layout

connortechnology, 2024-06-06:
On my 1920x1080 monitor which is fairly common size/aspect ratio, some improvements can be made to use of space, etc. 
Here are a few of my thoughts, please add your own.

The tags edit/add does not need to be 100% width.  I could be 100% of the left info column though.  This would give that vertical space back to the img, which would allow it to fill the horizontal space a bit better as well.

Event Id and Name rows could be merged in the info column
StartTime/EndTime could be merged into one line maybe
frames/alarm frames could be merged
the various scores could be merged
Storage and diskspace could be merged, storage could be a link to the storage area view
The under video status and dvr controls could be merged into 1 line, again giving more vertical space for video.

connortechnology, 2024-06-06:
@IgorA100 @Simpler1

IgorA100, 2024-06-06:
Oooohhh...
There is so much work here that I couldn’t even think what would look better where and how.
Thanks Isaac for starting your thoughts.

> StartTime/EndTime could be merged into one line maybe

The line will be 10 characters longer, and if the recording start day does not coincide with the recording end day, up to 18 characters longer. 18 is already a lot.

Other than that, I don't mind.

Simpler1, 2024-06-06:
1920x1080 is by far the most common screen resolution and that should be our priority, but we can't forget about narrow displays (mobile).

I also want to clean up more of the screens for mobile (width < 500px).  For example, the Options screen is completely useless at less than 500px.

My current cameras have an aspect ratio of 4:3.  Do we know what the most common camera aspect ratios are?

I agree that the tags can be moved to only take up the width of the properties area, but we need to make sure that it doesn't get removed for mobile (when the properties aren't displayed at all).  

I don't think we should combine multiple of the other properties into single rows.  That would make the rows too wide.

I also agree that the DVR controls and the video status can be combined into line line when it is no wider than the video display (not for mobile).

IgorA100, 2024-06-06:
> Do we know what the most common camera aspect ratios are?

Most of my cameras (and the cameras of my friends) are 16:9. Don't forget about portrait camera orientation!

> That would make the rows too wide.

Not everywhere.
I think this can be combined (suggested by Isaac):
- Event Id and Name rows could be merged in the info column
- frames/alarm frames could be merged
- the various scores could be merged
- Storage and diskspace could be merged

I completely agree about support for mobile devices, although I myself have not used a smartphone in principle for more than 10 years.

Aleksanderk123, 2024-11-30:
In the event list I think that merging Storage and DiskSpace into the same colcoumn would really free up some horizontal space for smaller devices. You can not sort by Storage anyways, so if you keep the sorting to DiskSpace, it would work pretty much the same as before.

connortechnology, 2024-12-01:
We aren't really discussing the events list here.  I wil make the storage column sortable. Not sure why it isn't.  You can select which columns are visible.