ZoneMinder/zoneminder #4077: Replace the top menu with a side menu

IgorA100, 2024-06-14:
I propose to discuss the option of adding a side menu like here: https://www.bootstrap.gallery/demos/milestone-admin-dashboard/index.html#

Sources: https://www.bootstrap.gallery/details/milestone---bootstrap-admin-panel

connortechnology, 2024-06-14:
Go for it, but it has to be configurable. Not everyone wants that.

baudneo, 2024-06-15:
Seems like a good idea, I do agree on configurable.

IgorA100, 2024-06-15:
Those. I have to add an option in settings to be able to use the current menu or a new one.
Right ?

baudneo, 2024-06-16:
> Those. I have to add an option in settings to be able to use the current menu or a new one. Right ?

yes

connortechnology, 2024-09-13:
I may actually get around to merging some code that would at least provide a sidebar if turned on.  I want to also make the top navbar really completely configurable in terms of content.

Anyways, going to re-open this and put it on the 1.39 roadmap.

IgorA100, 2024-09-13:
I wanted to implement this menu, but I haven't managed to structure the sections and levels of the menu in my head yet, and also understand what to display in the side menu and what in the top one.
Later I worked on the Montage page for viewing recorded events with the new Timeline and haven't finished it yet (there's really a lot of work and nuances...). Therefore, the question with the side menu remains open.
Perhaps it's worth discussing what sections and levels are needed in the menu?

IgorA100, 2025-03-04:
@connortechnology 

Here's an interesting option: https://github.com/azouaoui-med/pro-sidebar-template/tree/master
Demo: https://azouaoui-med.github.io/pro-sidebar-template
It seems to work fine on mobile devices too.
Take a look when you have time and tell me what you think.

connortechnology, 2025-03-04:
Looks fine.  No problem with it.  Side bars are not hard these days, lots of options.

Point Is, let's add it, but it has to be configurable.  Some people will want top menus as is, others may want a left menu.

IgorA100, 2025-03-04:
> Some people will want top menus as is, others may want a left menu.

That is, duplicate the information in the left and top menu?
And make it possible to switch between using the top/side menu

> Side bars are not hard these days, lots of options.

Yes, but many of them are either not very convenient, or buggy, or have complicated code....

connortechnology, 2025-03-04:
Yes.  I want to make the content in the header configurable as well.  But that's for 1.39+

IgorA100, 2025-03-04:
Ok, let's discuss.

IgorA100, 2025-03-09:
@connortechnology 
Just a mockup of the sidebar layout.
Isaac, what are your thoughts

[image]
[image]

IgorA100, 2025-03-10:
@connortechnology 
Menu is already filled. The layout is partially corrected.
Isaac, I need your opinion. What do you like and what do you not like.

[image]
[image]

IgorA100, 2025-03-10:
Mobile view
[image]
[image]

IgorA100, 2025-03-11:
I liked it in one row and with scrolling. It's convenient on a mobile device!

IgorA100, 2025-03-11:
Made the scroll bar prettier and increased the height of the status bar.
The scroll bar is specially made thin so as not to take up too much space and not to irritate. No one will cling to the scroll bar on a mobile device anyway, and the shift will be a grab on the status bar itself.
I like it...

[image]

IgorA100, 2025-03-21:
I decided to move the filters to the Side bar.
This is what I got:

There is still a lot of work to do...
I am waiting for criticism

[image]

[image]

baudneo, 2025-03-23:
I am loving it

IgorA100, 2025-03-27:
So, the work on the new menu is almost finished.
For touch screens, Swipe from left to right is available to display the menu.

The active menu item is highlighted.
The menu item for setting the current filter is always located at the top and does not move when scrolling other menu items. This menu item for displaying the current filter is currently implemented only for the pages: 'console', 'montage', 'montagereview', 'watch' and 'events'. Maybe for some other pages the filter should be moved to a pull-out block?

Please give your opinion on the menu items. The order of the items has been slightly changed and the Events page item has been added.

[image]
[image]
[image]
[image]

closed 2025-06-28 by connortechnology