ZoneMinder/zoneminder #4083: Erro when trying to install on archlinux

nipnipj, 2024-06-24:
The following error appears when trying to install on archlinux:
```
[ 24%] Linking CXX static library libRtspServer.a
[ 24%] Built target RtspServer
[ 25%] Building CXX object src/CMakeFiles/zm.dir/zm_analysis_thread.cpp.o
In file included from /home/cuysaurus/zoneminder/src/zoneminder-1.36.33/src/zm_monitor.h:24,
                 from /home/cuysaurus/zoneminder/src/zoneminder-1.36.33/src/zm_analysis_thread.cpp:3:
/home/cuysaurus/zoneminder/src/zoneminder-1.36.33/src/zm_camera.h: In member function ‘int Camera::getChannels()’:
/home/cuysaurus/zoneminder/src/zoneminder-1.36.33/src/zm_camera.h:111:51: error: ‘AVCodecParameters’ {aka ‘struct AVCodecParameters’} has no member named ‘channels’
  111 |     return mAudioStream ? mAudioStream->codecpar->channels : -1;
      |                                                   ^~~~~~~~
make[2]: *** [src/CMakeFiles/zm.dir/build.make:76: src/CMakeFiles/zm.dir/zm_analysis_thread.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:484: src/CMakeFiles/zm.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

synthead, 2024-06-25:
This seems related to https://github.com/ZoneMinder/zoneminder/issues/3983.

After upgrading on Arch, I get this:

```
$ ldd /usr/bin/zmc
	linux-vdso.so.1 (0x000072f32d9b2000)
	libz.so.1 => /usr/lib/libz.so.1 (0x000072f32d980000)
	libjpeg.so.8 => /usr/lib/libjpeg.so.8 (0x000072f32cb64000)
	libgnutls.so.30 => /usr/lib/libgnutls.so.30 (0x000072f32c96f000)
	libpcre.so.1 => /usr/lib/libpcre.so.1 (0x000072f32d905000)
	libgcrypt.so.20 => /usr/lib/libgcrypt.so.20 (0x000072f32c7de000)
	libmariadb.so.3 => /usr/lib/libmariadb.so.3 (0x000072f32c798000)
	libavformat.so.60 => not found
	libavcodec.so.60 => not found
	libavutil.so.58 => not found
	libswscale.so.7 => not found
	libswresample.so.4 => not found
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x000072f32c400000)
	libm.so.6 => /usr/lib/libm.so.6 (0x000072f32c6ad000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x000072f32c3d3000)
	libc.so.6 => /usr/lib/libc.so.6 (0x000072f32c1e7000)
	/lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x000072f32d9b4000)
	libp11-kit.so.0 => /usr/lib/libp11-kit.so.0 (0x000072f32c051000)
	libidn2.so.0 => /usr/lib/libidn2.so.0 (0x000072f32c68b000)
	libunistring.so.5 => /usr/lib/libunistring.so.5 (0x000072f32bea1000)
	libtasn1.so.6 => /usr/lib/libtasn1.so.6 (0x000072f32be8a000)
	libnettle.so.8 => /usr/lib/libnettle.so.8 (0x000072f32be33000)
	libhogweed.so.6 => /usr/lib/libhogweed.so.6 (0x000072f32bdea000)
	libgmp.so.10 => /usr/lib/libgmp.so.10 (0x000072f32bd44000)
	libgpg-error.so.0 => /usr/lib/libgpg-error.so.0 (0x000072f32bd1c000)
	libssl.so.3 => /usr/lib/libssl.so.3 (0x000072f32bc42000)
	libcrypto.so.3 => /usr/lib/libcrypto.so.3 (0x000072f32b600000)
	libffi.so.8 => /usr/lib/libffi.so.8 (0x000072f32bc37000)
```

connortechnology, 2024-06-25:
Your ffmpeg libraries are too new.  There are fixes in the release-1.36 branch

synthead, 2024-07-18:
If there are fixes available, can a new release be prioritized?  As of this writing, the only way to use ZoneMinder with Arch (aside from Docker, etc.) is to use downgraded packages.  It'd be great if this was seamless 🙂

connortechnology, 2024-10-22:
1.36.35 has been released. Reopen if necessary.

closed 2024-10-22 by connortechnology