ZoneMinder/zoneminder #4099: ZM stops listening to ONVIF events if a camera becomes unavailable and then reappears on the network.

IgorA100, 2024-07-22:
**Describe Your Environment**
- Version of ZoneMinder: v1.37.60

**To Reproduce**
Steps to reproduce the behavior:
1. Configure ZM and camera to listen to ONVIF events. SOAP WSA COMPLIANCE = disabled
2. Disconnect the camera from the network.
3. After some time, turn the camera back on to the network.
4. ZM no longer listens to ONVIF events.

After the error appears
`ONVIF Couldn't create subscription! HTTP Error, <html><body><h1>404 Not Found</h1></body></html> | zm_monitor.cpp | 1203`
ZM stops listening for ONVIF events and does not start listening again after the camera comes online.
```
 Error("ONVIF Couldn't create subscription! fault:%s, detail:%s", soap_fault_string(soap), detail ? detail : "null");
 _wsnt__Unsubscribe wsnt__Unsubscribe;
 _wsnt__UnsubscribeResponse wsnt__UnsubscribeResponse;
 proxyEvent.Unsubscribe(response.SubscriptionReference.Address, NULL, &wsnt__Unsubscribe, wsnt__UnsubscribeResponse);
 soap_destroy(soap);
 soap_end(soap);
 soap_free(soap);
 soap = nullptr;
```
I have not studied the code in detail, but it is probably necessary to first check whether the camera is available on the network, then check the availability of the subscription and generate an ONVIF listening error only if the camera itself is accessible and there are problems with listening to events.
If the camera is not available on the network, of course there will be problems with ONVIF. And is it necessary to report this?
Also, after the camera appears on the network, it is necessary to resume checking the ability to listen to ONVIF events

IgorA100, 2024-08-20:
@connortechnology 
Have you fixed this problem?

connortechnology, 2024-08-20:
No.

commit 9b9f0159 (2024-09-16): Move onvif event listener code to zm_monitor_onvif.cpp.  In the Poll thread, if listener is not healthy, restart it.  Fixes #4099
closed 2024-09-16 by connortechnology