ZoneMinder/zoneminder #4100: Segmentation fault: 'zmc -m 1' exited abnormally, exit status 255

ylexus, 2024-07-22:
**Describe Your Environment**
- Version of ZoneMinder: 1.37.63
- How you installed ZoneMinder: followed https://wiki.zoneminder.com/Debian_12_Bookworm_with_Zoneminder_1.36.33, "Debian 12 with Zoneminder 1.37.x"
- Full name and version of OS: Debian GNU/Linux 12 (bookworm), Linux rpi 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
- Browser name and version: N/A

**If the issue concerns a camera**
- Make and Model: Reolink E1 Zoom
- frame rate: 20
- resolution: 2560x1920
- ZoneMinder Source Type: rtsp (used the probe)

**Describe the bug**
It all worked fine yesterday. I have been configuring the system on my Raspberry Pi 5, but can only relate to this issue a random regular update and a change to the recording path in the filesystem. Suddenly, the recording stopped and the camera view just shows a single frame then pauses for a long time, then another frame. In the log I can see repeated crashes:
```
7/22/24, 7:18:18 PM GMT+1 | zmdc | 11571 | ERR | 'zmc -m 1' exited abnormally, exit status 255 | zmdc.pl | -
-- | -- | -- | -- | -- | -- | --
7/22/24, 7:18:18 PM GMT+1 | zmc_m1 | 16925 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 54
7/22/24, 7:18:18 PM GMT+1 | zmc_m1 | 16920 | INF | Kitchen Reolink: 50 - Capturing at 20.08 fps, capturing bandwidth 316589bytes/sec Analysing at 0.00 fps | zm_monitor.cpp | 1934
7/22/24, 7:18:16 PM GMT+1 | zmc_m1 | 16920 | INF | Kitchen Reolink: 20 - Capturing at 19.02 fps, capturing bandwidth 712853bytes/sec Analysing at 0.00 fps | zm_monitor.cpp | 1934
7/22/24, 7:18:15 PM GMT+1 | zmc_m1 | 16920 | INF | Kitchen Reolink: 0 - Capturing at 0.00 fps, capturing bandwidth 198bytes/sec Analysing at 0.00 fps | zm_monitor.cpp | 1934
7/22/24, 7:18:13 PM GMT+1 | zmc_m1 | 16920 | INF | Starting Capture version 1.37.63 | zmc.cpp | 221
7/22/24, 7:18:12 PM GMT+1 | zmdc | 11571 | INF | 'zmc -m 1' starting at 24/07/22 19:18:12, pid = 16920 | zmdc.pl | -
7/22/24, 7:18:12 PM GMT+1 | zmdc | 16920 | INF | 'zmc -m 1' started at 24/07/22 19:18:12 | zmdc.pl | -
7/22/24, 7:18:12 PM GMT+1 | zmdc | 11571 | INF | Starting pending process, zmc -m 1 | zmdc.pl | -
7/22/24, 7:17:42 PM GMT+1 | zmdc | 11571 | ERR | 'zmc -m 1' exited abnormally, exit status 255 | zmdc.pl | -
7/22/24, 7:17:42 PM GMT+1 | zmc_m1 | 16915 | FAT | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 52
7/22/24, 7:17:42 PM GMT+1 | zmc_m1 | 16915 | ERR | addr2line -Cfip -e /usr/bin/zmc 0x9d0c0 0x80224 0x4fc10 0x175348 | zm_signal.cpp | 133
7/22/24, 7:17:42 PM GMT+1 | zmc_m1 | 16915 | ERR | and execute the following command for more information: | zm_signal.cpp | 132
7/22/24, 7:17:42 PM GMT+1 | zmc_m1 | 16915 | ERR | Backtrace complete, please install debug symbols (typically zoneminder-dbg) | zm_signal.cpp | 131
7/22/24, 7:17:42 PM GMT+1 | zmc_m1 | 16915 | ERR | Backtrace 7: /lib/aarch64-linux-gnu/libc.so.6(+0xe7f9c) [0x7fffb7087f9c] | zm_signal.cpp | 112
7/22/24, 7:17:42 PM GMT+1 | zmc_m1 | 16915 | ERR | Backtrace 6: /lib/aarch64-linux-gnu/libc.so.6(+0x7ee58) [0x7fffb701ee58] | zm_signal.cpp | 112
7/22/24, 7:17:42 PM GMT+1 | zmc_m1 | 16915 | ERR | Backtrace 5: /lib/aarch64-linux-gnu/libstdc++.so.6(+0xce9dc) [0x7fffb730e9dc] | zm_signal.cpp | 112
7/22/24, 7:17:42 PM GMT+1 | zmc_m1 | 16915 | ERR | Backtrace 4: /usr/bin/zmc(+0x175348) [0x55561b985348] | zm_signal.cpp | 112
7/22/24, 7:17:42 PM GMT+1 | zmc_m1 | 16915 | ERR | Backtrace 3: /usr/bin/zmc(+0x4fc10) [0x55561b85fc10] | zm_signal.cpp | 112
7/22/24, 7:17:42 PM GMT+1 | zmc_m1 | 16915 | ERR | Backtrace 2: /usr/bin/zmc(+0x80224) [0x55561b890224] | zm_signal.cpp | 112
7/22/24, 7:17:42 PM GMT+1 | zmc_m1 | 16915 | ERR | Backtrace 1: linux-vdso.so.1(__kernel_rt_sigreturn+0) [0x7fffb93c07b0] | zm_signal.cpp | 112
7/22/24, 7:17:42 PM GMT+1 | zmc_m1 | 16915 | ERR | Backtrace 0: /usr/bin/zmc(+0x9d0c0) [0x55561b8ad0c0] | zm_signal.cpp | 112
7/22/24, 7:17:42 PM GMT+1 | zmc_m1 | 16915 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 54
7/22/24, 7:17:42 PM GMT+1 | zmc_m1 | 16912 | INF | Kitchen Reolink: 50 - Capturing at 19.52 fps, capturing bandwidth 302792bytes/sec Analysing at 0.00 fps | zm_monitor.cpp | 1934
7/22/24, 7:17:41 PM GMT+1 | zmc_m1 | 16912 | INF | Kitchen Reolink: 30 - Capturing at 20.00 fps, capturing bandwidth 754234bytes/sec Analysing at 0.00 fps | zm_monitor.cpp | 1934
7/22/24, 7:17:40 PM GMT+1 | zmc_m1 | 16912 | INF | Kitchen Reolink: 10 - Capturing at 3.10 fps, capturing bandwidth 48064bytes/sec Analysing at 0.00 fps | zm_monitor.cpp | 1934
7/22/24, 7:17:37 PM GMT+1 | zmc_m1 | 16912 | INF | Starting Capture version 1.37.63 | zmc.cpp | 221
7/22/24, 7:17:36 PM GMT+1 | zmdc | 11571 | INF | 'zmc -m 1' starting at 24/07/22 19:17:36, pid = 16912 | zmdc.pl | -
7/22/24, 7:17:36 PM GMT+1 | zmdc | 16912 | INF | 'zmc -m 1' started at 24/07/22 19:17:36 | zmdc.pl | -
7/22/24, 7:17:36 PM GMT+1 | zmdc | 11571 | INF | Starting pending process, zmc -m 1
```

**To Reproduce**
Steps to reproduce the behavior:
Just start the app, the error is in the log.

**Expected behavior**

**Debug Logs**
See above - I am not 100% sure how to enable DEBUG level. I tried enabling LOG_DEBUG, but that does not have any effect. Possibly I need to set multiple options. Please could you help me with this?

PVasileff, 2024-07-23:
I had the same problem. I see that today they have released an update that corrects this behavior:

`zoneminder/bookworm,now 1.37.63~20240723.45-bookworm amd64 [installed]`

ylexus, 2024-07-27:
Thanks, this is now fixed.

closed 2024-07-27 by ylexus