ZoneMinder/zoneminder #4101: When executing the CMD_PAUSE, CMD_SEEK, CMD_QUERY commands getCmdResponse stream error occurs: socket_sendto( /run/zm/zms-204049s.sock ) failed: Connection refused

IgorA100, 2024-07-22:
**Describe Your Environment**
- Version of ZoneMinder v1.37.60, v1.37.62

I've been researching for a long time and couldn't figure out why the "Connection refused" socket errors are happening in my code (new Montage page).
It turned out that this error also appears on the Event page, for example, when the “Pause” button is pressed.

There is also a nuance with the socket.
If you send the CMD_PAUSE command, and then regularly send the CMD_QUERY command, then after 5 minutes, the socket stops responding (Connection refused error).
I could not find where this time of 5 minutes is configured.

IgorA100, 2024-07-23:
The error does not appear on all recorded events, but only on some and, as a rule, closer to the end of the event.
Perhaps the event itself has problems with timestamps.
But I don't know how to identify the problem yet.

IgorA100, 2024-07-23:
So, the problematic files have a variable FPS, for example from 0.5 to 33.
But I can't figure out why the "Connection refused" error appears.
Perhaps the socket is closing prematurely, even though CMD_QUERY commands are being sent?
@connortechnology can you tell me something?

connortechnology, 2024-08-13:
When paused, we send an image (the same one) every 5 sec to keep the connection alive. We weren't doing that. I just fixed it.

commit 5237641e (2024-08-13): Fix calculation for when to send a keepalive image.  Possible fixes #4101
closed 2024-08-13 by connortechnology