ZoneMinder/zoneminder #4103: Signature by key ABE4C7F993453843F0AEB8154D0BF748776FFB04 uses weak algorithm (rsa1024)

markohrastovec, 2024-07-26:
**Describe Your Environment**
- Version of ZoneMinder 1.36.33+dfsg1-1build4 arm64
- How you installed ZoneMinder iconnor ppa
- Full name and version of OS Armbian 24.5.3 Noble

**Describe the bug**
When updating via apt there is always a warning

W: https://ppa.launchpadcontent.net/iconnor/zoneminder-1.36/ubuntu/dists/jammy/InRelease: Signature by key ABE4C7F993453843F0AEB8154D0BF748776FFB04 uses weak algorithm (rsa1024)

printed to stderr. It is not so annoying until one uses apticron. apticron is a utility that checks if new updates are available and notifies (vie e-mail) about updates available for the OS. apticron handles this as an error and an e-mail error report is sent every day regardless if there are updates available or not.

It is time to make ppa key stronger.

**To Reproduce**
Steps to reproduce the behavior:
1. Install apticron.
2. Error about zoneminder's weak algrotihm is sent via e-mail every day.

**Expected behavior**
An e-mail should be sent only when new packages are available.

**Debug Logs**
NA

connortechnology, 2024-08-13:
https://askubuntu.com/questions/1510042/how-to-fix-launchpad-ppa-which-uses-a-key-with-weak-algorithm

Not sure that there is anything I can do about it.  The packages were last generated over a year ago, so my releasing 1.36.34 yesterday may fix this by causing launchpad to update the InRelease file.  We will see.

SteveGilvarry, 2024-09-13:
https://discourse.ubuntu.com/t/new-requirements-for-apt-repository-signing-in-24-04/42854#a-ppa-on-launchpad-triggers-a-weak-algorithm-warning-what-do-i-do
"PPAs are currently in the process of being upgraded to a 4096-bit RSA key and we expect that upgrade to be complete by release time. No action is needed (or possible) from PPA owners."

closed 2024-09-13 by SteveGilvarry