ZoneMinder/zoneminder #4109: Cameras showing offline when they are still recording

daniel-hodgson, 2024-08-05:
**Describe Your Environment**
- 1.37.63
- PPA
- Ubuntu 24.04
- Firefox 128.0.3

**If the issue concerns a camera**
- Varies
- 11fps
- 720p
- FFMPEG

**Describe the bug**
All cameras in the system report offline, even though they are actively recording footage. FPS Report Interval is set to default (100). Reducing the interval to 50 or even 22 does not fix the issue. `Monitors` table in the DB also showed `Function` column was set to `Monitor` instead of `Record` for all cams. Updating this manually in the DB to `Record` did not resolve this issue.

This seems to have started after a recent power outage, and ZM is running inside of a virtual machine (RHEL 7, QEMU-KVM).

**Expected behavior**
All cameras should show online.

connortechnology, 2024-08-06:
The Function column is no longer used.  Monitor status is contained in the Monitor_Status table.

A failed db update could be causing the problem in which case there will be errors logged.  Check your logs.

dghodgson, 2024-08-06:
False alarm! It was a Timezone mismatch. Relevant error was in the web_php log.

closed 2024-08-13 by connortechnology