ZoneMinder/zoneminder #4110: Error message in logs when trying to generate video file from cam thta only captures JPEGs.

monohydroxy, 2024-08-06:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
- Full name and version of OS
- Browser name and version (if this is an issue with the web interface)
- v1.36.34
- Installed from PPA - https://ppa.launchpadcontent.net/iconnor/zoneminder-proposed/ubuntu noble InRelease
- Ubuntu 24.04 - Server version (no GUI)
- Firefox - v128.0.3

**If the issue concerns a camera**
- Make and Model
- frame rate
- resolution
- ZoneMinder Source Type:
- This issue does not concern a camera.

**Describe the bug**
A clear and concise description of what the bug is.
I have one camera setup to only capture JPEGs and no video (mp4/H265/etc) so to turn this into a video, I go to the 'view=video' page by clicking on a 'Generate Video' button.  On that 'view=video' page, there is another Generate Video button.  When I click that button there is a log entry with the error: 

- web_php		2362605	ERR	actions can no longer be performed without POST. Requested: video for request   /usr/share/zoneminder/www/index.php	200

The URL of this page is:
- .../zm/index.php?view=video&eid=1636

The web page says that it created the file, but when I click on 'download' I get a file not found error.

**To Reproduce**
Steps to reproduce the behavior:
1. Using Montage review or something similar to get to a single event for a camera that only has JPEGs captured.
2. Click on the Generate Video Button (looks like a video camera on a piece of paper with the corner turned down).
3. On that new page click on a similar Generate Video Button.
4. Switch to the System Log page.
5. Look at the most recent error for 'actions can no longer be performed without POST' error.
6. Try to download the generated video.
7. See error message about file not found (or sometimes an empty file).

**Expected behavior**
A clear and concise description of what you expected to happen.
Expect that the generate video button will do a POST and the video will be generated successfully.  Expect that the video can then be downloaded.

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

I'm not sure what debug logs would be useful here.

```

connortechnology, 2024-08-08:
Please try latest update.  The reported error should not happen anymore.  I don't think it is an actual fix, as I was unable to reproduce the file not found aspect of this report.  We do need debug level logs, from the _web_php area.  So under Options->Logging, set DEBUG to checked and in LOG_DEBUG_TARGET put _web.  There should be a file in /var/log/zm/web_php.log containing the debug logs.

monohydroxy, 2024-08-10:
I updated today and the POST error is fixed.  I still get the file not found error.  The URL is:
- http://192.168.0.195/zm/index.php?view=video&eid=18948&downloadIndex=0

I can find the file manually here:
- /mnt/REDACTED/zoneminder/5/2024-08-09/18948/Event-18948-r1-s1.mp4

Note that I'm using a custom (non-default) location for file storage.  I turned on the debug option and clicked on Download again and it downloaded a file with these contents:

<br />
<b>Fatal error</b>:  Allowed memory size of 134217728 bytes exhausted (tried to allocate 65028096 bytes) in <b>/usr/share/zoneminder/www/skins/classic/views/video.php</b> on line <b>96</b><br />

The file is 215Mbytes.  Is there another log that you need?

--MH

connortechnology, 2024-08-10:
Aha!  That error explains it all!  We have to turn off buffering when sending files.  I'll look into that.
In the meantime, you could increase the amount of ram available to php.  Edit your php.ini  in /etc/php/php8.1/apache2/php.ini and look for the memory setting.  It defaults to 128Mb which is really low.

commit 7aa473b8 (2024-08-11): Handle more than one level of output buffering when cleaning and ending them so we can send the video file so we don't run out of ram.  Fixes #4110
closed 2024-08-11 by connortechnology