ZoneMinder/zoneminder #4111: ppa source no longer available ? (fresh install)

koopernick, 2024-08-08:
I'm trying to install Zoneminder on Ubuntu 22.04 (new install), following your documentation => https://zoneminder.readthedocs.io/en/stable/installationguide/ubuntu.html

Here is the error message when trying to install your source 

add-apt-repository ppa:iconnor/zoneminder-1.36

```
[...]
Ign:6 https://ppa.launchpadcontent.net/iconnor/zoneminder-1.36/ubuntu noble InRelease
Err:7 https://ppa.launchpadcontent.net/iconnor/zoneminder-1.36/ubuntu noble Release
  404  Not Found [IP: 185.125.190.80 443]
Reading package lists... Done
E: The repository 'https://ppa.launchpadcontent.net/iconnor/zoneminder-1.36/ubuntu noble Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Both of your docker source are depreciated
=> https://github.com/dlandon/zoneminder.machine.learning
=> https://github.com/dlandon/zoneminder.master-docker

What's the official and working way to install Zoneminder on Ubuntu 22.04 jammy LTS ? 

Thank you.

connortechnology, 2024-08-08:
You are NOT installing on 22.04. You are installing on 24.04.  

If you want to stay with 1.36, use our proposed release ppa ppa:iconnor/zoneminder-proposed or use our development version at ppa:iconnor/zoneminder-master

koopernick, 2024-08-09:
Ho, my bad. I most likely picked the wrong iso...

Thanks for the fast answer !

closed 2024-08-09 by koopernick