ZoneMinder/zoneminder #4112: Building 1.34 still fails with FFmpeg 7.0

Nocifer, 2024-08-12:
**Describe Your Environment**
- Version of ZoneMinder: 1.34
- How you installed ZoneMinder: from source
- OS: Arch Linux

**Describe the bug**
The build fails with:
```
[ 77%] Building CXX object src/CMakeFiles/zm.dir/zm_regexp.cpp.o
In file included from /build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg.cpp:22:
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg.cpp: In function ‘void zm_dump_stream_format(AVFormatContext*, int, int, int)’:
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg.cpp:338:32: error: ‘AVCodecParameters’ {aka ‘struct AVCodecParameters’} has no member named ‘channels’
  338 |         codec->profile, codec->channels, codec->sample_rate);
      |                                ^~~~~~~~
/build/zoneminder/src/zoneminder-1.36.34/src/zm_logger.h:200:72: note: in definition of macro ‘logPrintf’
  200 |       Logger::fetch()->logPrint(false, __FILE__, __LINE__, logLevel, ##params); \
      |                                                                        ^~~~~~
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg.cpp:337:5: note: in expansion of macro ‘Debug’
  337 |     Debug(1, "profile %d channels %d sample_rate %d",
      |     ^~~~~
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg.cpp: In function ‘void zm_free_codec(AVCodecContext**)’:
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg.cpp:596:18: warning: ‘int avcodec_close(AVCodecContext*)’ is deprecated [-Wdeprecated-declarations]
  596 |     avcodec_close(*ctx);
      |     ~~~~~~~~~~~~~^~~~~~
In file included from /build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg.h:127,
                 from /build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg.cpp:20:
/usr/include/libavcodec/avcodec.h:2387:5: note: declared here
 2387 | int avcodec_close(AVCodecContext *avctx);
      |     ^~~~~~~~~~~~~
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_input.cpp: In member function ‘int FFmpeg_Input::Close()’:
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_input.cpp:146:20: warning: ‘int avcodec_close(AVCodecContext*)’ is deprecated [-Wdeprecated-declarations]
  146 |       avcodec_close(streams[i].context);
      |       ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
In file included from /build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_input.h:12,
                 from /build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_input.cpp:1:
/usr/include/libavcodec/avcodec.h:2387:5: note: declared here
 2387 | int avcodec_close(AVCodecContext *avctx);
      |     ^~~~~~~~~~~~~
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_input.cpp: In member function ‘AVFrame* FFmpeg_Input::get_frame(int)’:
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_input.cpp:170:17: warning: ‘void av_init_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
  170 |   av_init_packet(&packet);
      |   ~~~~~~~~~~~~~~^~~~~~~~~
In file included from /usr/include/libavcodec/codec_par.h:33,
                 from /usr/include/libavformat/avformat.h:312,
                 from /build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_input.h:10:
/usr/include/libavcodec/packet.h:653:6: note: declared here
  653 | void av_init_packet(AVPacket *pkt);
      |      ^~~~~~~~~~~~~~
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_input.cpp: In member function ‘AVFrame* FFmpeg_Input::get_frame(int, double)’:
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_input.cpp:303:30: error: ‘AVFrame’ {aka ‘struct AVFrame’} has no member named ‘pkt_duration’; did you mean ‘duration’?
  303 |   if (frame->pts + 10*frame->pkt_duration < seek_target) {
      |                              ^~~~~~~~~~~~
      |                              duration
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_input.cpp:316:42: error: ‘AVFrame’ {aka ‘struct AVFrame’} has no member named ‘pkt_duration’; did you mean ‘duration’?
  316 |     while (frame && (frame->pts + frame->pkt_duration < seek_target)) {
      |                                          ^~~~~~~~~~~~
      |                                          duration
make[2]: *** [src/CMakeFiles/zm.dir/build.make:468: src/CMakeFiles/zm.dir/zm_ffmpeg.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [src/CMakeFiles/zm.dir/build.make:496: src/CMakeFiles/zm.dir/zm_ffmpeg_input.cpp.o] Error 1
/build/zoneminder/src/zoneminder-1.36.34/src/zm_mpeg.cpp: In destructor ‘VideoStream::~VideoStream()’:
/build/zoneminder/src/zoneminder-1.36.34/src/zm_mpeg.cpp:510:30: warning: ‘int avcodec_close(AVCodecContext*)’ is deprecated [-Wdeprecated-declarations]
  510 |                 avcodec_close( codec_context );
      |                 ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
In file included from /build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg.h:127,
                 from /build/zoneminder/src/zoneminder-1.36.34/src/zm_mpeg.h:23,
                 from /build/zoneminder/src/zoneminder-1.36.34/src/zm_mpeg.cpp:20:
/usr/include/libavcodec/avcodec.h:2387:5: note: declared here
 2387 | int avcodec_close(AVCodecContext *avctx);
      |     ^~~~~~~~~~~~~
/build/zoneminder/src/zoneminder-1.36.34/src/zm_mpeg.cpp: In member function ‘double VideoStream::ActuallyEncodeFrame(const uint8_t*, int, bool, unsigned int)’:
/build/zoneminder/src/zoneminder-1.36.34/src/zm_mpeg.cpp:613:23: warning: ‘void av_init_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
  613 |         av_init_packet( pkt );
      |         ~~~~~~~~~~~~~~^~~~~~~
In file included from /usr/include/libavcodec/avcodec.h:44:
/usr/include/libavcodec/packet.h:653:6: note: declared here
  653 | void av_init_packet(AVPacket *pkt);
      |      ^~~~~~~~~~~~~~
/build/zoneminder/src/zoneminder-1.36.34/src/zm_packet.cpp: In constructor ‘ZMPacket::ZMPacket()’:
/build/zoneminder/src/zoneminder-1.36.34/src/zm_packet.cpp:45:17: warning: ‘void av_init_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
   45 |   av_init_packet(&packet);
      |   ~~~~~~~~~~~~~~^~~~~~~~~
In file included from /usr/include/libavcodec/avcodec.h:44,
                 from /build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg.h:127,
                 from /build/zoneminder/src/zoneminder-1.36.34/src/zm_packet.h:23,
                 from /build/zoneminder/src/zoneminder-1.36.34/src/zm_packet.cpp:19:
/usr/include/libavcodec/packet.h:653:6: note: declared here
  653 | void av_init_packet(AVPacket *pkt);
      |      ^~~~~~~~~~~~~~
/build/zoneminder/src/zoneminder-1.36.34/src/zm_packet.cpp: In constructor ‘ZMPacket::ZMPacket(Image*, const timeval&)’:
/build/zoneminder/src/zoneminder-1.36.34/src/zm_packet.cpp:65:17: warning: ‘void av_init_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
   65 |   av_init_packet(&packet);
      |   ~~~~~~~~~~~~~~^~~~~~~~~
/usr/include/libavcodec/packet.h:653:6: note: declared here
  653 | void av_init_packet(AVPacket *pkt);
      |      ^~~~~~~~~~~~~~
/build/zoneminder/src/zoneminder-1.36.34/src/zm_packet.cpp: In copy constructor ‘ZMPacket::ZMPacket(ZMPacket&)’:
/build/zoneminder/src/zoneminder-1.36.34/src/zm_packet.cpp:85:17: warning: ‘void av_init_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
   85 |   av_init_packet(&packet);
      |   ~~~~~~~~~~~~~~^~~~~~~~~
/usr/include/libavcodec/packet.h:653:6: note: declared here
  653 | void av_init_packet(AVPacket *pkt);
      |      ^~~~~~~~~~~~~~
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_camera.cpp: In member function ‘virtual int FfmpegCamera::Close()’:
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_camera.cpp:592:18: warning: ‘int avcodec_close(AVCodecContext*)’ is deprecated [-Wdeprecated-declarations]
  592 |     avcodec_close(mVideoCodecContext);
      |     ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
In file included from /build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg.h:127,
                 from /build/zoneminder/src/zoneminder-1.36.34/src/zm_image.h:23,
                 from /build/zoneminder/src/zoneminder-1.36.34/src/zm_camera.h:23,
                 from /build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_camera.h:23,
                 from /build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_camera.cpp:20:
/usr/include/libavcodec/avcodec.h:2387:5: note: declared here
 2387 | int avcodec_close(AVCodecContext *avctx);
      |     ^~~~~~~~~~~~~
/build/zoneminder/src/zoneminder-1.36.34/src/zm_ffmpeg_camera.cpp:599:18: warning: ‘int avcodec_close(AVCodecContext*)’ is deprecated [-Wdeprecated-declarations]
  599 |     avcodec_close(mAudioCodecContext);
      |     ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/usr/include/libavcodec/avcodec.h:2387:5: note: declared here
 2387 | int avcodec_close(AVCodecContext *avctx);
      |     ^~~~~~~~~~~~~
make[1]: *** [CMakeFiles/Makefile2:484: src/CMakeFiles/zm.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

SteveGilvarry, 2024-09-13:
@Nocifer Changes for FFmpeg 7 have been back ported to https://github.com/ZoneMinder/zoneminder/tree/release-1.36 for testing to be released as 1.36.35. This will be the last 1.36 release as we move to preparing 1.38 for release. Please close if you are happy with this or if you want to wait for the release to be available and tested let us know

Nocifer, 2024-09-16:
Hello, and sorry for the belated reply. I'm more than happy that you have a 1.36.35 release in the works that will tackle the remaining FFmpeg 7.0 issues, and happier still that you'll be finally moving forward with 1.38 after that.

Thank you for your work; I'm closing this as fixed.

closed 2024-09-16 by Nocifer