ZoneMinder/zoneminder #4116: Filter errors

Rickcovell, 2024-08-17:
**Describe Your Environment**
- Version of ZoneMinder 1.36.33
- How you installed ZoneMinder from source
- Full name and version of OS Truenas core 13.2

all filters are exiting abnormally with exit status 2 
Excerpt from the logs

"8/17/24, 8:02:47 PM UTC",zmdc,,42017,ERR,"'zmfilter.pl --filter_id=4 --daemon' exited abnormally, exit status 2",zmdc.pl,'-
"8/17/24, 8:02:46 PM UTC",zmdc,,42143,INF,"'zmfilter.pl --filter_id=4 --daemon' started at 24/08/17 21:02:46",zmdc.pl,'-
"8/17/24, 8:02:46 PM UTC",zmdc,,42017,INF,"'zmfilter.pl --filter_id=4 --daemon' starting at 24/08/17 21:02:46, pid = 42143",zmdc.pl,'-
"8/17/24, 8:02:46 PM UTC",zmdc,,42017,INF,"Starting pending process, zmfilter.pl --filter_id=4 --daemon",zmdc.pl,'-
"8/17/24, 8:02:43 PM UTC",zmdc,,42017,ERR,"'zmfilter.pl --filter_id=1 --daemon' exited abnormally, exit status 2",zmdc.pl,'-
"8/17/24, 8:02:43 PM UTC",zmdc,,42017,ERR,"'zmfilter.pl --filter_id=2 --daemon' exited abnormally, exit status 2",zmdc.pl,'-
"8/17/24, 8:02:43 PM UTC",zmc_m4,,334692,INF,"Front Door: 100 - Capturing at 4.08 fps, capturing bandwidth 107934bytes/sec Analysing at 4.08 fps",zm_monitor.cpp,1681
"8/17/24, 8:02:43 PM UTC",zmdc,,42017,INF,"'zmfilter.pl --filter_id=1 --daemon' starting at 24/08/17 21:02:43, pid = 42136",zmdc.pl,'-
"8/17/24, 8:02:43 PM UTC",zmdc,,42136,INF,"'zmfilter.pl --filter_id=1 --daemon' started at 24/08/17 21:02:43",zmdc.pl,'-
"8/17/24, 8:02:43 PM UTC",zmdc,,42017,INF,"Starting pending process, zmfilter.pl --filter_id=1 --daemon",zmdc.pl,'-

bit of a newbie to this so no idea what i'm doing, or what ZM is doing. if this can be solved I'll need a method to clear logs as the number is ever growing. 
Thanks Rick

connortechnology, 2024-08-17:
Try running it from the command line and see what it says.

sudo zmfilter.pl --filter_id=1

Rickcovell, 2024-08-17:
[image]
or
Cant locate MIME/Entity.pm in @INC (you may need to Install the MIME::Entity module) (@INC contains: /usr/local/lib/perl5/site_perl/mach/5.36 /usr/local/lib/perl5/site_perl /usr/local/lib/perl5/5.36/mach /usr/local/lib/perl5/5.36) at /usr/local/bin/zmfilter.pl line 146

any idea?

Rick

connortechnology, 2024-08-17:
Options->Email->New Mail Modules .
CHeck it.  Click save.

Rickcovell, 2024-08-18:
Thanks for that, nearly worked, still got errors, but opting out of e-mails altogether solved the problem. obviously more work to do in understanding the e-mail setup.

connortechnology, 2024-08-20:
You need to install the relevant perl modules for sending mail.   I don't know what's available under truenas, but I suppose there is always CPAN.

Rickcovell, 2024-08-20:
Oh there's a whole new world of knowledge to be had. Thanks i'll keep picking at it and see what happens

closed 2024-08-20 by Rickcovell