ZoneMinder/zoneminder #4119: Log contains errors "...Out of range value for column 'MaxScore' at row 1".

monohydroxy, 2024-08-21:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
- Full name and version of OS
- Browser name and version (if this is an issue with the web interface)
- v1.36.34 (up-to-date from PPA  as of August 20, 2024)
- Installed from PPA - https://ppa.launchpadcontent.net/iconnor/zoneminder-proposed/ubuntu noble InRelease
- Ubuntu 24.04 - Server version (no GUI)
- Firefox - v128.0.3

**If the issue concerns a camera**
- Make and Model
- frame rate
- resolution
- ZoneMinder Source Type:
- The issue does not seem to be source dependent.

**Describe the bug**
A clear and concise description of what the bug is.
The log files contain the following error:
- Can't run query UPDATE Events SET Name='Event-29930', EndDateTime = from_unixtime(1724194075), Length = 9027.19, Frames = 0, AlarmFrames = 0, TotScore = 0, AvgScore = 0, MaxScore = -1 WHERE Id = 29930 AND Name='New Event': Out of range value for column 'MaxScore' at row 1
It seems like MaxScore used to be a signed int and was changed to an unsigned int in the database.  I'm not sure why or when these errors occur.  The logs has about 5-15 in a second, then nothing for hours.  I don't think there is any user interaction when the error messages are generated.  There is a gap in the recording at the same time as the error.  I'm not sure if they are related.

**To Reproduce**
Steps to reproduce the behavior:
1. I don't know how to reproduce the issue, I just wait for it to occur, then review the logs.

**Expected behavior**
A clear and concise description of what you expected to happen.
I expected  no error message to be generated, assuming I setup my cameras correctly and assuming the camera is working correctly.

**Debug Logs**
What logs would be helpful?
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

garrettgee2001, 2024-08-24:
Experiencing the same issue.  Ubuntu 22.04.  Seems to only be happening on a set of 4K "linked" cameras.  Can't figure out how to reproduce, but does seem to happen when the camera that is the "linked" one creates an event, the linked camera throws this error in ZM.

connortechnology, 2024-09-18:
2899a4debc245e77ff6bad5cacecc26c7b1a0154 should address this.  New proposed packages building. version should show 1.36.35.

connortechnology, 2024-10-22:
1.36.35 has been released which should address it.  Reopen if needed.

closed 2024-10-22 by connortechnology