ZoneMinder/zoneminder #4125: Annoying ask to update stored credentials when saving monitor

connortechnology, 2024-08-27:
I've tried to solve this one and so far failed.  The browser asks about saving credentials when saving a monitor, or even just browsing away.  Makes for an extra annoying click.

IgorA100, 2024-09-15:
Hmm, I haven't seen that. Strange.
Can I get a video so I can check it step by step?

Aleksanderk123, 2024-12-04:
I get this sometimes as well. I don't know how to reproduce it, so it's hard to trace.
If I remember correctly, this was a bigger problem in earlier versions, right?

@IgorA100 It acts the same as when you log into a website, and it asks you to save/update your credentials for that site.

Maybe it has something to do with the names for ONVIF or RTSP credentials? I will look into it, and see if I can find something out.

IgorA100, 2024-12-05:
> I get this sometimes as well.

Once a couple of times, about a year ago, I think I had this behavior.
But for at least half a year there have been no problems with this. Although I have not fixed anything.
In general, more detailed information is needed to reproduce this situation