ZoneMinder/zoneminder #4127: GridStack slow to init when a lot of monitors on montage

connortechnology, 2024-09-03:
With over 70 monitors, it can take over 40 seconds before first render.  firefox will complain about gridstack init slowing down the browser, etc.

I notice that we are calling init twice, which could perhaps be improved.  Not sure what else can be done.

IgorA100, 2024-09-13:
I'll definitely watch it, but later.
I don't have time right now :(

IgorA100, 2024-09-13:
> With over 70 monitors, it can take over 40 seconds before first render

1. Can you make a video of the whole rendering process?
2. Does the rendering speed depend on the selected "Layout"? I think "8Wide" will load faster than "3Wide"

connortechnology, 2024-09-13:
https://github.com/user-attachments/assets/4ecf6c90-9580-4ede-af1c-857b46ea22aa

connortechnology, 2024-09-13:
Yeah 8 Wide is faster.

IgorA100, 2024-09-13:
> I notice that we are calling init twice

GridStack initialization consumes almost no resources. Yes, when opening a page, initialization occurs twice, this is normal for now.

Thanks for the video.

As far as I understand, the long page rendering occurs due to loading a large amount of information.
Reducing "Stream quality" to -50 should speed up page rendering.

IgorA100, 2024-09-14:
Maybe I found the reason.

connortechnology, 2024-10-22:
It is massively improved, while faster would be even better, I think we can close this.

closed 2024-10-22 by connortechnology