ZoneMinder/zoneminder #4132: Fullscreen mode not persistent (replay/Gapless Events)

BlueMax, 2024-09-14:
**Describe Your Environment**
zoneminder/jammy,now 1.37.63~20240906093428-jammy amd64
ppa.launchpadcontent.net/iconnor/zoneminder-master/ubuntu/ jammy main
Firefox 130

**Describe the bug**
Event view:
When enabling 'replay/Gapless Events' mode and switching to 'fullscreen' (button) it jumps back to non-fullscreen on next event (after event ends). It should keep fullscreen mode until user exits it on purpose.

connortechnology, 2024-09-14:
Unfrotunately there is no way to maintain fullscreen between page reloads.  So what we have to do here is NOT do a page reload.

IgorA100, 2024-09-17:
If you hide the Stats display to the left of the video, then the full-screen mode works correctly, because the page does not reload.
Of course, you need to do it so that the page does NOT reload when Stats is displayed

BlueMax, 2024-09-18:
Stats are always hidden here. Page reloads are still active.

BlueMax, 2024-09-18:
Mjpeg has bad quality/efficiency. I wouldn't trade that for the problem.

connortechnology, 2024-10-22:
Have we finished addressing this?  Can we close it?

BlueMax, 2024-10-23:
The problem is still there. Same page reloads as before.

IgorA100, 2024-10-23:
This problem requires deeper analysis and probably AJAX requests should be used.