ZoneMinder/zoneminder #4134: video.js progress bar broken by panzoom

connortechnology, 2024-09-16:
5e76dfe273691e0d4d521e3ed643430ab7169a23 caused clicking on the progress bar provided by video.j.s to stop working.

IgorA100, 2024-09-17:
> to stop working.

progress bar not working?

I once noticed that the progress bar does not work, i.e. the strip does not move, but I do not think that this could be related to PabZoom.

I have now rolled back the changes https://github.com/ZoneMinder/zoneminder/commit/5e76dfe273691e0d4d521e3ed643430ab7169a23 and the progress bar still does not work when using video.js

I will try to figure it out.

connortechnology, 2024-09-17:
When I say doesn't work, I mean that when you click on it, seeking does not happen.  This is the bar provided by video.js, not the one with the ticks below.

IgorA100, 2024-09-17:
> This is the bar provided by video.js,

I got it.
But even after rolling back https://github.com/ZoneMinder/zoneminder/commit/5e76dfe273691e0d4d521e3ed643430ab7169a23 the progress bar generated by video.js doesn't work for me. Everything is very strange.....

I have "Live" displayed next to the progress bar. Maybe that's the problem? That is, video.js doesn't understand that a recording is being played and broadcasts in real time? That's why the timeline doesn't move.
I haven't studied video.js, but it seems to me that panzoom couldn't break the progress bar generated by video.js
I'll try to figure out what caused the breakdown.
[image]

> not the one with the ticks below

After applying #4136 this panel should work. If you click on this progress bar, then the transition to the desired time occurs.

connortechnology, 2024-09-17:
That's weird.  Doesn't do that for me.  I suspect you are correct, I can see why it wouldn't have a progress bar if it was live.

For me, simply commenting out the panzoom div is enough to make the clicking/seeking work.  It's like the panzoom div is overlaying the progress bar, but I don't see why that would be, and any attempts to play with z-index etc have no effect.

IgorA100, 2024-09-17:
In my case, `var duration = this.techGet('duration');` is not defined correctly in the video.js file
For some reason, my duration is always 0. I haven't figured this out yet.

I was able to simulate your situation.
I will try to solve this problem.

IgorA100, 2024-09-17:
Conflict interception of events panzoom & video.js

PR #4137 (2024-09-17): Fix: Disabled preventDefault() when initializing PanZoom (panzoom.js)
Fix  #4134
PR #4136 (2024-09-18): Feat: Progress bar now works when viewing MP4 via video.js on Event page
I'm not sure if this PR fixes #4134
But now the Progress bar correctly displays the timeline..
I don't know if this ever worked before....
[image]
closed 2024-09-17 by connortechnology