ZoneMinder/zoneminder #4146: Console reports some cameras not running, but they are running and recording

rnveach, 2024-09-21:
**Describe Your Environment**
- Version of ZoneMinder: 1.36.34 
- How you installed ZoneMinder: `ppa:iconnor/zoneminder-1.36`
- Full name and version of OS: Ubuntu 22.04.5 LTS
- Browser name and version: Firefox 130.0.1

**If the issue concerns a camera**
- Not a concern with the camera itself

**Describe the bug**
I went into the ZM console today and noticed it is reporting 2 of my cameras (1 and 5) are not running. However, I can still view the cameras through the web interface, confirmed they are still live footages, and confirmed it is still recording those cameras when events occur. The web interface of the camera says `Monitor is not capturing. We will be unable to provide an image`, but it does provide an image.

I ran `zmu -l` and it reports the cameras have framerates though the console doesn't report any.
````
WARNING: MYSQL_OPT_RECONNECT is deprecated and will be removed in a future version.
  Id Func State TrgState    LastImgTim RdIdx WrIdx LastEvt FrmRate
   1    3     4        0 1726936466.87337634     0 1727750    5.31
   2    3     1        0 1726936466.96334100     0 1727310    7.51
   3    3     1        0 1726936467.14349784     1 1727568   10.02
   5    3     1        0 1726936467.00907484     1 1727719    8.40
````

I have not tried a reboot to see if it fixes it. The cameras may be similar, but they all have different brands and models (Wansview, Agasio, etc..). All are on wifi. The issue hasn't gone away in the last hour.

Console Image:
[image]

4 is blackened because it is intentionally not running to avoid any confusion.

**To Reproduce**
Steps to reproduce the behavior:

Unknown. I don't view the console that often. Maybe some software was upgraded and caused a disruption?

**Expected behavior**
I expect the web interface, zmu, and backend to behave identical. If the backend for the camera is functioning properly and has no issues working, I would expect the web interface to show the same status and not incorrectly report the camera as not running.

**Debug Logs**
No JS errors besides `The Components object is deprecated. It will soon be removed`.

The only errors in the console for camera 1 is:
````
9/21/24, 12:00:09 AM EDT	zmc_m1		703121	ERR	Select error: Interrupted system call	zm_remote_camera_http.cpp	241
9/21/24, 12:00:09 AM EDT	zmc_m1		703121	ERR	Unable to read subheader	zm_remote_camera_http.cpp	937
9/21/24, 12:00:09 AM EDT	zmc_m1		703121	ERR	Unable to get response, disconnecting	zm_remote_camera_http.cpp	1087
9/21/24, 12:00:09 AM EDT	zmc_m1		703121	ERR	Failed to capture image from monitor 1 Monitor-1 (1/1)	zmc.cpp	305
````

There are a bunch of warnings like:
````
9/21/24, 12:32:17 PM EDT	zmc_m1		743499	WAR	Found iterator at beginning of queue. Some thread isn't keeping up
````

Otherwise most of the displays for the single camera is just informational, and it is printing out informational like its FPS and bandwidth. Camera 5 has similar logs.

connortechnology, 2024-09-21:
Your FPS report interval is large.  THis issue is already fixed in the upcoming 1.36.35 which you can get from the proposed ppa.  Or for now just reduce your fps report interval down to something lower than 600.

rnveach, 2024-09-21:
Thank you. I am unsure which option you mean for `fps report interval`. I assume this isn't MAX_RESTART_DELAY?

SteveGilvarry, 2024-09-22:
https://zoneminder.readthedocs.io/en/stable/userguide/definemonitor/definemonitor_misc.html#misc-tab
I believe default is 100, FPS Report Int/FPS = Updates to console values in seconds. So 600/5 would be updating every 2mins.