ZoneMinder/zoneminder #4151: Areas incorrect (alarm pixels outside area)

BlueMax, 2024-09-26:
**Describe Your Environment**
zoneminder/jammy,now 1.37.63~20240924172342-jammy amd64
ppa.launchpadcontent.net/iconnor/zoneminder-master/ubuntu/ jammy main
Firefox 130.0.1
720p h264 passthrough rtsp 20fps (5fps analysis, greyscale)

**Describe the bug**
Analysis frames show alarm blobs outside any area. Looks like there's some shifting/offset.

[image]  [image]

BlueMax, 2024-09-26:
Looks like the 'inactive' zone disconnects the red area/polygons and the points will be reconnected directly.
This is a camera day/night switch (all pixel alarm).

[image]

AJ0070, 2026-08-29:
This looks fixed by 83ed47abf. The scan-line fill was stepping one active edge at a time instead of in pairs, so it filled the gaps between the arms of a non-convex polygon, which is the mask that gates alarm pixels.

closed 2026-08-30 by connortechnology