ZoneMinder/zoneminder #4155: Backported schema update from 1.37.25 to 1.36.34 won't work on some versions of mysql databases

itsTurnip, 2024-09-27:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
1.36.34
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
[zoneminder-containers image](https://github.com/zoneminder-containers/zoneminder-base)
- Full name and version of OS
Debian 12 (as for image base)

**Describe the bug**
Upgrading from 1.36.33 to 1.36.34 will fail with some old versions of databases. The following error will appear for mariadb 5.5.68 which is packed on CentOS 7.

**To Reproduce**
Run upgrade from 1.36.33 to 1.36.34 with mysql version < 5.6.5.

**Expected behavior**
Successfull migration from 1.36.33 to 1.36.34

**Debug Logs**
```
Upgrading DB to 1.36.34 from 1.36.33
ERROR 1067 (42000) at line 10: Invalid default value for 'UpdatedOn'
```

itsTurnip, 2024-09-27:
I know it's kind of an old database which is not likely to be supported anymore, but I think 1.36.* version should not drop this support.

For anyone who is looking for a solution: you can just add column `UpdateOn` to `MonitorStatus` table and then create a [trigger](https://stackoverflow.com/a/20017760/12866133) to fill it out.

connortechnology, 2024-10-22:
The fix was included in 1.36.35.  I think we can close this.

commit 7cc4ac2f (2024-09-27): Remove default of NOW from UpdatedOn in Monitor_Status field because old mysql can't handle it.  Explicilty set it in zmc.
Fixes #4155
closed 2024-10-22 by connortechnology