ZoneMinder/zoneminder #4162: zoneminder 1.37.64 option to switch between camera fucntions/modes missing.

kvacha13, 2024-10-02:
**Describe Your Environment**
- Version of ZoneMinder v1.37.64 
- How you installed ZoneMinder :
 I upgraded from 1.36 to 1.37.64
 Upgrade from ZM 1.36 to 1.37 according Post by [iconnor](https://forums.zoneminder.com/memberlist.php?mode=viewprofile&u=17112) » Tue Aug 08, 2023 2:45 pm

sudo add-apt-repository ppa:iconnor/zoneminder-master
sudo apt install zoneminder

No need to remove it first. Make sure you take a db backup first.
Top
  
- Full name and version of OS
  Linux my-arm64-server 5.15.0-1068-oracle #74~20.04.1-Ubuntu SMP Tue Sep 3 09:58:50 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
- Browser name and version (if this is an issue with the web interface)
  chrome Version 129.0.6668.72 and firefox 130.0.1 (64-bit)

**Describe the bug**
1) the switch option between functions/modes (monitor, modect...) is missing. In previous versions, it was possible by clicking the function hyperlink, but in 1.37.64 there is no such option (attaching picture).
[image]
2) Force alarm button is not triggering alarm even if the camera is in modect mode..

connortechnology, 2024-10-02:
Function has been replaced with Recording and Analysing settings.

kvacha13, 2024-10-02:
OK. and how to switch on/off recording mode (modect) to monitoring mode? how can I stop recording mode?

connortechnology, 2024-10-22:
The popup functionality may come back, I want to do a better job of it.  Console is getting a rework.

The Force Alarm button should work again. The logic behind all that was weird... I think it is better now.

kvacha13, 2024-10-22:
I've just noted that the latest release version on Github is v1.36.35 while I've a Zoneminder of v 1.37.64. Have i missed something?

kvacha13, 2024-11-03:
I noticed the following strange behavior:
I have two cameras in Zoneminder with the same function settings (both: Capturing, Analyzing Always, Recording: OnMotion). But in zmNinja, those two cameras have different Mode statuses: one has Modect, and another has Monitor. Both cameras are recording but the one with Mode:Monitor records all the time instead of recording on Motion (and does not send push notifications). and I cannot manage to set it to record onMotion (on Modect mode). Previously it was easy to do from zoneminder (dropdown function list) and also from zmNinja. 
Can you @connortechnology help with the issue? Is there a manual on how to set RecordingAlways, RecordingOnMOtion, JustMonitoring, etc... ?

[image]
[image]