ZoneMinder/zoneminder #4176: Zoneminder 1.36.34 FTBS on Fedora Rawhide (41)

knight-of-ni, 2024-10-14:
Hey Isaac - I hope things are going well. 

I am on the verge of leaving town for work, so I have not gone through any investigation into the following build issues, nor have I checked if these have been fixed in master. Just documenting these here, to be followed up on when one of us has time to look into it.

Looks like a recent change to openssl is tripping up jwt. I wonder if a newer jwt exists, that fixes this:
```
/builddir/build/BUILD/zoneminder-1.36.34-build/zoneminder-1.36.34/dep/jwt-cpp/include/jwt-cpp/jwt.h:916:53: warning: ‘int EC_KEY_check_key(const EC_KEY*)’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  916 |                                 if (EC_KEY_check_key(pkey.get()) == 0) throw ecdsa_exception(error::ecdsa_error::invalid_key);
      |                                     ~~~~~~~~~~~~~~~~^~~~~~~~~~~~
```

RPMFusion just upgraded for ffmpeg 7.0.2. We've got just one API change to deal with:

```
In file included from /builddir/build/BUILD/zoneminder-1.36.34-build/zoneminder-1.36.34/src/zm_ffmpeg.cpp:22:
/builddir/build/BUILD/zoneminder-1.36.34-build/zoneminder-1.36.34/src/zm_ffmpeg.cpp: In function ‘void zm_dump_stream_format(AVFormatContext*, int, int, int)’:
/builddir/build/BUILD/zoneminder-1.36.34-build/zoneminder-1.36.34/src/zm_ffmpeg.cpp:338:32: error: ‘AVCodecParameters’ {aka ‘struct AVCodecParameters’} has no member named ‘channels’
  338 |         codec->profile, codec->channels, codec->sample_rate);
      |                                ^~~~~~~~
```

Full build log is here if you need it:
https://gist.github.com/knight-of-ni/41f4c19f3f71d52d948f2eaf52f3cb41

connortechnology, 2024-10-15:
I think the ffmpeg issues are fixed in the release-1.36 branch, ready to go for 1.36.35.  

In master, we moved to a MUCH newer jwt-cpp 0.6.  I see that there has been a 0.7 release. will upgrade master and if that goes ok, I'd be ok with updating 1.36 as well.

knight-of-ni, 2024-10-21:
I began working on this. After looking more closely, I realized the jwt problem was just a warning.

Focusing on fixing the ffmpeg API problem has snowballed into additional API changes that didn't show up into I fixed the first.
The changes are spread out across more source files than I anticipated. I cannot solve this with a simple patch to the 1.36.34 source file. I think the better solution at this point will be to wait until 1.36.35 release. 

Do you have an eta for 1.36.35?

connortechnology, 2024-10-21:
Are you sure the ffmpeg stuff hasn't already been addressed?  Please look in the release-1.36 branch, it is all ready to go for release, am just waiting on this issue.  If the jwt stuff is just a warning then I can release today.  But now I'd like confirmation that the current pre-release code works.  I do NOT want to do a 1.36.36

knight-of-ni, 2024-10-21:
Yes, zoneminder builds successfully from the release-1.36 branch against Fedora rawhide w/ ffmpeg 7.

I was trying to give you more time, by patching the existing 1.36.34 tarball, to correct the ffmpeg related build errors. Unfortunately, that turned out not to be as simple as I had hoped.

If you are ready to release now, however, that would certainly resolve this.

closed 2024-10-22 by knight-of-ni