ZoneMinder/zoneminder #4177: Errors with debian-sys-maint mysql connections

GregOriol, 2024-10-23:
**Describe Your Environment**
- Version of ZoneMinder : 1.36.35
- How you installed ZoneMinder : ppa iconnor
- Full name and version of OS : Ubuntu 22.04.5 LTS
- Browser name and version : not a web issue

**Describe the bug**
In the system logs, I can see errors like:
```
10/23/24, 9:32:43 AM UTC	zmstats		2619653	ERR	The user specified as a definer ('debian-sys-maint'@'localhost') does not exist	zmstats.pl	-
...
10/23/24, 9:20:40 AM UTC	web_php		3545729	ERR	SQL-ERR 'SQLSTATE[HY000]: General error: 1449 The user specified as a definer ('debian-sys-maint'@'localhost') does not exist', statement was 'DELETE FROM Events WHERE Id = ?' params:13333	includes/database.php	161
```

However, I can't find where this debian-sys-maint@localhost comes from: I have a configuration file /etc/zm/zm.conf defining ZM_DB_TYPE, ZM_DB_HOST, ZM_DB_NAME, ZM_DB_USER, ZM_DB_PASS and they are not set to this.

**Expected behavior**
This seems right now to prevent zm to connect to mysql

I've search the source code of the project and couldn't find any reference to such a user for the database.

Where could this come from?

GregOriol, 2024-10-23:
This was due to mysql triggers on some zm tables, so not in the code. They were created by the debian-sys-maint user for some reason and not the configured zm user, and this user has been deleted recently on our database. This created the error.

The solution I found was to use phpmyadmin, find all the tables having triggers with this wrong DEFINER (in `information_scheme`.`TRIGGERS`) then go to each table's triggers tab and edit the definer field to set the correct user.

closed 2024-10-23 by GregOriol

connortechnology, 2024-10-23:
Yeah deleting the debian-sys-maint user causes troubles for sure.