ZoneMinder/zoneminder #4178: Motion detection, crashes, detection stops in 1.37

IgorA100, 2024-10-23:
**Describe Your Environment**
- Version of ZoneMinder 1,37 updated 22-10-24
- How you installed ZoneMinder [Isaac Connor's PPA]
- Full name and version of OS - Ubuntu 24.04.1

**If the issue concerns a camera**
- China
- frame rate - 15-20 (I haven't tried other values)
- resolution - 2048*1536 H264B

**Detection area settings**
Alarm Check Method: Blobs
Min/Max Pixel Threshold (0-255): 10
Filter Width/Height (pixels): 3 - 15
Zone Area: 100%
Min/Max Alarmed Area: 0.12%
Min/Max Filtered Area: 0.12%
Min/Max Blob Area: 0.12%
Min/Max Blobs: 1

I tried changing Alarmed Area, Filtered Area, Blob Area, but this only affects the size of the detection area. The problem remains.

**Describe the bug**
Initially, detection works correctly, but after recording from 2 to 10 events, the camera starts recording continuously and stops analyzing movements. Then it can stop recording altogether! If you go to the camera settings and simply re-save, the camera works correctly for some time.

No errors. Free memory ~30Gb out of 64Gb, CPU load does not exceed 40%

I understand that there is little information, but if you run debugging level 3, there will be a huge file.
I have not been able to solve this problem for several weeks, it is probably a bug.

connortechnology, 2024-10-23:
Not going to be able to help without a debug log.  No one else is experiencing this.

IgorA100, 2024-10-24:
After disabling ONVIF on the camera, there are no problems so far (for 20 hours). It is possible that ONVIF events transmitted from the camera interfere with the analysis of ZM detection.
Debug level 3 for 2 hours is 500 MB

IgorA100, 2024-10-25:
So, today's behavior:
During the day, there was definitely movement near the camera, but no motion detection was detected for 4 hours by ZM. Then the detection started working again.
I have a level 3 debug file of 1.2 GB. What should I do?

connortechnology, 2024-10-25:
You could try https://cloud.zoneminder.com/upload/upload_center.html

Or I can setup and ftp account

IgorA100, 2024-10-25:
The archived file is lightweight.
Here is the link: https://cloud.kentavr.com.ru/s/NSFWCGNASofiLWa/download/_debug.zip

IgorA100, 2024-10-25:
There were definitely movements between these two events that were not detected by ZM
2024-10-25 14:59:37 | 2024-10-25 15:00:14
2024-10-25 19:00:57 | 2024-10-25 19:06:43

connortechnology, 2024-10-25:
Not sure if this is the problem yet, but because you are analysing at 5fps but only decoding keyframes... you may skip over keyframes and generally not achieve any analysis.  If you are going to only decode keyframes, then you really should not set an analysis limit so that it will analyze every keyframe.

IgorA100, 2024-10-25:
Now I completely stopped understanding what the "Analysis FPS" parameter means when "Decoding = Keyframes + Ondemand"
I assumed the following algorithm:
- **each** Keyframe is decoded, as well as 5 more frames during a second.

That is, if the camera broadcasts with FPS=20 and Keyframes are transmitted every 2 seconds, then it will be like this:
- The first key frame is analyzed
- 4 frames are skipped and the next frame is analyzed (and so on five times during a second or ten times during two seconds)
- The second key frame is analyzed
And so on.

Am I wrong in my reasoning?

connortechnology, 2024-10-25:
Decoding=Keyframes means only keyframes are decoded.  So you may skip over the only frame that was decoded.

If a frame was not decoded then we can't analyze it.

IgorA100, 2024-10-25:
So, in theory, the "Analysis FPS" parameter should not affect the analysis at all when Decoding=Keyframes?
But in fact it does affect?

connortechnology, 2024-10-25:
Ideally.  I have a patch where if decoding=KEYFRAMES and the packet is decoded it will analyse it, but that's not good enough.  analysis fps could be set to 1/s but keyframe interval could be 2/s and so then we would be analysing at 2/s.

IgorA100, 2024-10-26:
With the "Analysis FPS" parameter unset (empty), events started to be recorded constantly (but there were also event skips!!!). More precisely, this is what happened:
- I cleared the parameter at night.
- There was no movement and ZM did not record events, everything is OK.
- Movements appeared in the morning ~approximately at 2024-10-26 08:39:16, but ZM did not detect them!!!
- The next movement occurred at 2024-10-26 09:02:09 and ZM started recording an event. But after the movement ended, the event recording did not stop. The recording lasted 10 minutes instead of 15-20 seconds (there was movement only in the first 12 seconds)
```
Cause Motion:All
Notes Motion: All
Start Time 10/26/24, 9:02:09AM GMT+3
End Time 10/26/24, 9:12:09AM GMT+3
Duration 00:10:00
Frames 10802 Alarm:1
Score Total:6 Max:6 Avg:6
```
- The next events are:
```
Cause Continuous
Notes
Start Time 10/26/24, 9:12:09AM GMT+3
End Time 10/26/24, 9:22:09AM GMT+3
Duration 00:10:00
Frames 10799 Alarm:0
Score Total:0 Max:0 Avg:0
```
Although they shouldn't be there, because there was no movement....

Log: https://cloud.kentavr.com.ru/s/CKgyFiDZTdei2p7/download/_debug_02133.zip
That is, it got even worse......

connortechnology, 2024-10-26:
Oh... that's a problem.  I'll fix it.