ZoneMinder/zoneminder #4181: Permission denied to push

regularRandom, 2024-10-28:
Hello,

I am trying to create a PR to fix one small issue with encoding of emails but cannot push any change:

ERROR: Permission to ZoneMinder/zoneminder.git denied to regularRandom.
fatal: Could not read from remote repository.

I can push to my own repositories using my new SSH key but not into the ZoneMinder.

I cloned repo by SSH. Please help.

**Describe Your Environment**

1.36.34
**If the issue concerns a camera**

No
**Describe the bug**

Cannot push to ZoneMinder repository

**Expected behavior**

Successful push and PR

**Debug Logs**
```

<!--
<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>
-->

```

connortechnology, 2024-10-28:
You can't push to our fork.  You push to your own and then create a PR to ours.

connortechnology, 2024-11-27:
Feel free to simply email me about the fix. isaac@zoneminder.com