ZoneMinder/zoneminder #4186: Event playback skips some events

j-marz, 2024-11-15:
**Describe Your Environment**
- v1.37.50 installed from PPA
- Ubuntu 20.04.6 LTS
- Firefox v130.0.1

**Describe the bug**
Playback of filtered events misses some events if the the next sequential event ID has a timestamp that is earlier than the next event ID. Changing the event list sorting doesn't resolve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. cause a zmtrigger event across multiple monitors
2. configure different pre-event buffers per monitor (this changes the event start time - ie. larger pre-event buffer will have earlier start time even though it was trigger later in the zmtrigger list)
3. list matching events using a filter
4. sort events by ID, name or start time (issue occurs in all scenarios)
5. playback in ascending order
6. notice some events are skipped and never playback

**Expected behavior**
event playback honors the filter sorting and doesn't skip events

**Debug Logs**
NA

j-marz, 2024-11-15:
Looks like playback uses the filter query and doesn't update the query when the event list sorting changes (e.g. change table sorting in event list view after running filter and before playback). Updating the event filter to sort by ID instead of start time seems to work (ie. doesn't skip events), but the side effect is the events don't technically play in chronological order.

connortechnology, 2024-11-15:
in zms when it goes to get the next event, it uses the following SQL:
SELECT `Id` FROM `Events` WHERE `MonitorId` = %d AND `Id` > last_event_id ORDER BY `Id` ASC LIMIT 1
So yeah, if the Ids are out of sync, it will mess up. It ignores the filter sort.  In fact the filter is never passed to zms.  

Which view is this? event view, or montage review?

j-marz, 2024-11-16:
`view=events`
The filter is definitely used for the event view playback - changing `Sort by` in the filter results in `sort_field` changing in the event playback URL and therefore `skip_next`.

connortechnology, 2024-11-27:
What have you got for Options->Web->WEB_EVENT_SORT_FIELD ? I bet it's Id.  Should be changed to StartDateTime.

closed 2024-11-27 by connortechnology

j-marz, 2024-11-27:
It's already `StartDateTime`, maybe i need to change that to `Id`?
[image]

Sorting IS being applied via the filter. What isn't happening is after running a filter (to find events of interest) and then resorting the the events table, the re-sorted event table view is ignored in playback and the original filter query (used to produce the events list) is passed to the playback. I had to change the sort field in the event filter to `Id` to avoid missing events.
[image]

connortechnology, 2024-11-27:
Nah there was some code in there that made it always use E.Id.  Should be fixed now.  You'll have to update.

j-marz, 2024-12-06:
@connortechnology thanks for looking into this. I upgraded to v1.37.65 and still running into missed events during playback.

For example event ID 6056855 is skipped during playback. I assume because 6056855 has the same start time as 6056854
[image]

Before upgrading, i was able to get all events to play if i sorted the event filter by ID. The only issue was some events wouldn't play in chronological order, but it was a technicality and not a blocker because in my case it was <1 min out for some events.

After, upgrade, it doesn't matter if i change the sort field in the filter or and events table, 6056855 is always skipped during playback of events. 

Can you please take another look or revert the changes in https://github.com/ZoneMinder/zoneminder/commit/359330b3840676ff2f5e032d4f0889f5db06360c ?

j-marz, 2024-12-06:
Changing Options->Web->WEB_EVENT_SORT_FIELD also doesn't help.

connortechnology, 2024-12-09:
Try latest.  We were using a > but your skipped event has the same starttime.. so I've changed it to >=

connortechnology, 2024-12-20:
Sweet!

commit 359330b3 (2024-11-27): When loading prev/next event, sort by StartDateTime, not Id. Id's are not guaranteed to be time-wise sequential.  Maybe Fixes #4186
commit ffeec4cd (2024-12-09): Add E.Id != and use >= and <= when comparing StartDateTime so that we don't skip events with the same startDateTime. Fixes #4186