ZoneMinder/zoneminder #4187: Does ZoneMinder support Gentoo?

glibg10b, 2024-11-25:
https://cloud.zoneminder.com/downloads.html has this:

[image]

However, executing that command doesn't work because ZoneMinder hasn't been in the official repos since June 2020.

closed 2024-11-27 by connortechnology

dekeonus, 2025-03-20:
@glibg10b there are builds in oubliette overlay.
```console
$ eselect repository enable oubliette
$ emaint sync -r oubliette
```
gentoo official dropped zoneminder a long time ago (last version there was 1.24 or there abouts)

While oubliette is not my overlay I am contributing to that overlay.

I am working to bring up 1.37.xx (and the ES + ML) , but it's a struggle with my older hardware: working out if my issues are not-quite-stable code or the machine crying (it's usually both).
1.36.35 is running on a x86 host (and thats  **real**  32bit metal not a 64bit machine with 32bit kernel and userspace - I did say older hardware). Though that 32bit host is just to see the build it's just not fast enough to do any analysis.
I am hoping to get a 1.37.xx build working with it so it can at least be a monitor via janus-gateway.

I don't run nginx so how it's working there *shrug*, hopefully someone tells me.

connortechnology, 2025-03-20:
Havn't tested on 32bit in a long while.  Otherwise ZMon gentoo shouldn't be a problem. Biggest issue we have is keeping up with ffmpeg