ZoneMinder/zoneminder #4195: ONVIF Cameras without stop/close event fail to trigger recording after first event closed

NothingTooSerious, 2024-12-13:
First up, thank you for ZM. The improvements in 1.37 branch is significant.

I looked into this problem and possibly identified the issue and logic fix. Not being familiar with ZM code base though, I am hesitant to produce a pull request until a review of the logic and impact assessment.

**Environment**

* Tested with a camera that only generates 'start' ONVIF motion events
* ZoneMinder version 1.37.65 commit 15dfaa5

**Expected**
Whenever ZM receives a start ONVIF Motion Event it will trigger or continue recording. Recording will stop from ZM normal stop recording settings.

**Observed**
ZM starts the initial recording then stops the initial recording. Any additional start ONVIF Motion Events fail to trigger any further recordings.

**Evidence**
The logs below demonstrate this.

Start the camera. The first ONVIF motion alarm triggers a recording as noted by the Triggered Start Event on ONVIF log.

```log
INF-zmc.cpp/221 [Starting Capture version 1.37.65]
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
INF-zm_monitor_onvif.cpp/208 [Triggered Start Event on ONVIF]
INF-zm_monitor.cpp/2179 [Slate: 1286 - Gone into alarm state PreAlarmCount: 0 > AlarmFrameCount:1 Cause:ONVIF]
INF-zm_monitor.cpp/2367 [Slate: 1286 - Opening event timestamp 1733853846 % 246]
INF-zm_monitor.cpp/2369 [Opened new event 13 ONVIF]
INF-zm_monitor.cpp/2211 [Slate: 1256 - Gone into alert state]
INF-zm_monitor.cpp/2216 [Slate: 1261 - Left alert state]
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
INF-zm_monitor.cpp/2344 [Slate: 2134 - Closing event 13 1733853875 - 1733853845 = 30 >= 600]
```

After this, logs are just the following repeated ad nauseam. There is no further triggering of recording.

```log
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
```

**Explaining why**

There are two source files involved being zm_monitor.cpp and zm_monitor_onvif.cpp.

In zm_monitor_onvif.cpp void Monitor::ONVIF::WaitForMessage() a motionAlert event triggers the following code on line 202:

```cpp
  } else {
[... 145 lines cut ...]
INF-zm_monitor.cpp/2369 [Opened new event 893 ONVIF]
INF-zm_monitor.cpp/2211 [Slate: 1517581 - Gone into alert state]
INF-zm_monitor.cpp/2216 [Slate: 1517586 - Left alert state]
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
INF-zm_monitor_onvif.cpp/215 [Triggered Start Event on ONVIF]
INF-zm_monitor.cpp/2179 [Slate: 1518061 - Gone into alarm state PreAlarmCount: 0 > AlarmFrameCount:1 Cause:ONVIF]
INF-zm_monitor.cpp/2211 [Slate: 1518031 - Gone into alert state]
INF-zm_monitor.cpp/2216 [Slate: 1518036 - Left alert state]
INF-zm_monitor_onvif.cpp/186 [ONVIF Got Motion Alarm! tns1:VideoSource/MotionAlarm State]
INF-zm_monitor_onvif.cpp/204 [Triggered Start on ONVIF]
INF-zm_monitor_onvif.cpp/215 [Triggered Start Event on ONVIF]
INF-zm_monitor.cpp/2179 [Slate: 1518525 - Gone into alarm state PreAlarmCount: 0 > AlarmFrameCount:1 Cause:ONVIF]
INF-zm_monitor.cpp/2211 [Slate: 1518495 - Gone into alert state]
INF-zm_monitor.cpp/2216 [Slate: 1518500 - Left alert state]
INF-zm_monitor.cpp/2344 [Slate: 1521094 - Closing event 893 1734083586 - 1734083466 = 120 >= 600]
```

Three recording events successfully triggered. Much better outcome. The alarms not being erased seems a non issue and does not leak memory.

So the logic appears incorrect as demonstrated by the test code.

I am running the test code change for a few days now without observing any issue.

I hope the above code review assists in developing a proper fix.

NothingTooSerious, 2025-02-24:
Is there anything else I could do to have this issue advance?

connortechnology, 2025-02-24:
Can you turn on debug please? Level 1 will do.

It seems to me, we should get an OK, but empty SOAP response, in which case we should become unalarmed.

connortechnology, 2025-02-25:
What do you think of removing the code in zm_monitor, and using the following around line 227:

Debug(1, "ONVIF polling : Got Good Response! %i", result);
      if (!tev__PullMessagesResponse.wsnt__NotificationMessage.size()) {
        if (!parent->Event_Poller_Closes_Event and alarmed) {
          alarmed = false;
          alarms.clear();
        }
      }

NothingTooSerious, 2025-02-25:
I will need to revisit the code and walk through its logic and impact from proposed changes.

Should be able to do it in next day and a half.

NothingTooSerious, 2025-02-26:
Level 1 debug logs attached as requested.
This was generated with zoneminder commit 52e68b0
Several event detections were successfully triggered and stopped.

[zm_onvif_debug_level1.log.gz](https://github.com/user-attachments/files/18989064/zm_onvif_debug_level1.log.gz)

NothingTooSerious, 2025-02-26:
Reviewed the code change proposed and tested it.

Identified some potential issues:

1. Does the proposed code on zm_monitor_onvif.cpp 222 code introduces a race condition?

It appears a new ONVIF event received could deactivate ONVIF alarm before the ONVIF triggered start event is raised in zm_monitor.cpp line 1957 causing a valid alarm to be missed.

Line 1958 of zm_monitor.cpp references onvif->isAlarmed(). This is updated independently by the new code introduced starting at line 222 of zm_monitor_onvif.cpp introducing the race condition between Alarm event and ONVIF event reception.

2. Alarmed state needs another non stop type ONVIF message to be received before it is cancelled.

3. Seeing Alarm frame counts and total score significantly higher since using this version.

Before: Alarm frames 6 - 20 and score 50 - 108.
After: Alarm frames 1400 - 4000 and score 12000 to 35388

This is for 2 - 3 minute duration events.

connortechnology, 2025-02-26:
1. Yes there is a race condition, but it was always there... we should get the lock in isAlarmed, and get it immediately after message reception.
2. Not necessarily, the poll has a timeout (20s).  My assumption is that if there isn't any events before the timeout occurs, then there isn't any motion, hence the transition to unalarmed.
3. I would expect so.  We were unalarming almost instantly.  Now we should be unalarming after 20s.

SteveGilvarry, 2026-01-02:
@connortechnology confirming that https://github.com/ZoneMinder/zoneminder/commit/97f92800dba4c9f263acdee46d32a82750b719c3 addressed this.

connortechnology, 2026-01-02:
I believe so.