ZoneMinder/zoneminder #4197: On demand capture after a while the camera shows offline

caffelgithub, 2024-12-14:
**Describe Your Environment**
- V1.37.65
- How you installed ZoneMinder: ppa 1.36 then upgraded to 1.37
- Full name and version of OS: Ubuntu 22.04 LTS
- Browser name and version: Firefox 133.0.3 (64-bit)

**If the issue concerns a camera**
- 2 Yoosee Camera (RTSP UDP), 1 V380 (RTSP TCP), Android Phone (RTSP TCP)
- 15 frame rate
- 1280x720p

**Describe the bug**
On demand capture after a while the camera appears offline in the UI but still records with a short duration of 5-10 seconds, after that a new event is created with a short duration too. it hape

I read in the forum that the issue is due to a timezone problem but I have tried to fix it to make sure the timezone is correct in zoneminder, in apache, in the ubuntu system, then in mysql, I use the timezone GMT+7 Asia Jakarta, but still no luck.

I also read that the issue is due to a high fps interval, but I checked one by one the camera fps interval is already default at 100. this problem only occurs when using the "on demand" capture setting but when using the "always" capture setting this problem does not occur.

**Logs**
"14/12/24 11.39.03 WIB",zmc_m2,23436,INF,"Cam-2 (Pintu Utama): image_count:638 - Closing event 2119, shutting down",zm_monitor.cpp,3415

[image]
[image]

avg-I, 2026-10-05:
I don't have logs handy right now, but I've just stumbled upon this issue and I can add two things.

If a camera is configured with:
- Capturing: On Demand
- Decoding: On Demand
- Recording: On Demand
then it becomes offline after a while.

Also, recording by an external trigger is not reliable.
Often, no recording starts and I guess that it could be related to the camera being "offline".

The idea of this configuration was that recording is triggered externally, so there is no need to put any load on camera otherwise.

I had to set Capturing: Always to work around both issues.

IgorA100, 2026-10-05:
@avg-I
Which version of ZM are you using?
It is recommended to test with version 1.39.36 from the Master branch.

avg-I, 2026-10-05:
@IgorA100 that was with version 1.38.3.
Let me see how easy/hard it is to try master on FreeBSD.