ZoneMinder/zoneminder #4207: Fails to build on Arm64

coreybruce, 2025-01-26:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit] 1.36.35-1
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc] AUR
- Full name and version of OS manjaro Arm64
- Browser name and version (if this is an issue with the web interface)

Hey I was told to report the issue by the maintainer of the AUR packages for zoneminder  zoneminder-git because it fails to compile on Arm64

https://aur.archlinux.org/packages/zoneminder

````
[ 16%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/net/TcpServer.cpp.o
[ 17%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/net/TcpSocket.cpp.o
[ 17%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/net/Timer.cpp.o
[ 18%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/net/Timestamp.cpp.o
[ 19%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/AACSource.cpp.o
[ 19%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/DigestAuthenticator.cpp.o
[ 20%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/G711ASource.cpp.o
[ 21%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/H264Parser.cpp.o
[ 21%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/H264Source.cpp.o
[ 22%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/H265Source.cpp.o
[ 23%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/MediaSession.cpp.o
[ 23%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/RtpConnection.cpp.o
[ 24%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/RtspConnection.cpp.o
[ 25%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/RtspMessage.cpp.o
[ 25%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/RtspPusher.cpp.o
[ 26%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/RtspServer.cpp.o
[ 27%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/VP8Source.cpp.o
[ 27%] Linking CXX static library libRtspServer.a
[ 27%] Built target RtspServer
make: *** [Makefile:136: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

connortechnology, 2025-01-27:
Thanks for the report. I can try to reproduce, but we have been building on arm64 for quite some time.... Unfortunately your report doesn't include the line that actually says what failed. Can you try to scroll and find it or post the whole output?

connortechnology, 2025-01-27:
Ok, our arm builds are failing too, but I don't think it's our fault. Please check to see if it's the same problem there:
Setting up python3-markupsafe (1.1.0-1build2) ...
Segmentation fault (core dumped)
Traceback (most recent call last):
  File "/usr/bin/py3compile", line 290, in <module>
    main()
  File "/usr/bin/py3compile", line 269, in main
    compile(files, versions,
  File "/usr/bin/py3compile", line 187, in compile
    pipe.send(fn)
  File "/usr/bin/py3compile", line 138, in py_compile
    stdin.write(filename.encode('utf-8') + b'\n')
BrokenPipeError: [Errno 32] Broken pipe
dpkg: error processing package python3-markupsafe (--configure):

coreybruce, 2025-01-27:
I am not having that same issue sadly

```
Building zoneminder...
==> Making package: zoneminder 1.36.35-1 (Tue 28 Jan 2025 07:49:28)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found 1.36.35.tar.gz
  -> Found 1.0-zm.tar.gz
  -> Found v3.2.0.tar.gz
  -> Found 055d81fe1293429e496b19104a9ed3360755a440.zip
  -> Found zoneminder-nginx.conf
  -> Found zoneminder-httpd.conf
  -> Found zoneminder-php.ini
  -> Found fcgiwrap-multiwatch.service
==> Validating source files with sha256sums...
    1.36.35.tar.gz ... Passed
    1.0-zm.tar.gz ... Passed
    v3.2.0.tar.gz ... Passed
    055d81fe1293429e496b19104a9ed3360755a440.zip ... Passed
    zoneminder-nginx.conf ... Passed
    zoneminder-httpd.conf ... Passed
    zoneminder-php.ini ... Passed
    fcgiwrap-multiwatch.service ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Extracting 1.36.35.tar.gz with bsdtar
  -> Extracting 1.0-zm.tar.gz with bsdtar
  -> Extracting v3.2.0.tar.gz with bsdtar
  -> Extracting 055d81fe1293429e496b19104a9ed3360755a440.zip with bsdtar
==> Starting prepare()...
==> Removing existing $pkgdir/ directory...
==> Starting build()...
CMake Deprecation Warning at CMakeLists.txt:5 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- The C compiler identification is GNU 14.1.1
-- The CXX compiler identification is GNU 14.1.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detected compiler: /usr/bin/cc
-- Performing Test HAVE_SENDFILE4_SUPPORT
-- Performing Test HAVE_SENDFILE4_SUPPORT - Success
-- Sendfile support: Linux/Solaris sendfile()
-- Building man pages: Yes (default)
-- Looking for libv4l1-videodev.h
-- Looking for libv4l1-videodev.h - found
-- Looking for linux/videodev2.h
-- Looking for linux/videodev2.h - found
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for backtrace
-- Looking for backtrace - found
-- Looking for backtrace_symbols
-- Looking for backtrace_symbols - found
-- Looking for ucontext.h
[... 162 lines cut ...]
[ 15%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/net/TaskScheduler.cpp.o
[ 15%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/net/TcpConnection.cpp.o
[ 16%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/net/TcpServer.cpp.o
[ 17%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/net/TcpSocket.cpp.o
[ 17%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/net/Timer.cpp.o
[ 18%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/net/Timestamp.cpp.o
[ 19%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/AACSource.cpp.o
[ 19%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/DigestAuthenticator.cpp.o
[ 20%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/G711ASource.cpp.o
[ 21%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/H264Parser.cpp.o
[ 21%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/H264Source.cpp.o
[ 22%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/H265Source.cpp.o
[ 23%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/MediaSession.cpp.o
[ 23%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/RtpConnection.cpp.o
[ 24%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/RtspPusher.cpp.o
[ 24%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/RtspMessage.cpp.o
[ 25%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/RtspConnection.cpp.o
[ 26%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/RtspServer.cpp.o
[ 27%] Building CXX object dep/RtspServer/CMakeFiles/RtspServer.dir/src/xop/VP8Source.cpp.o
[ 27%] Linking CXX static library libRtspServer.a
[ 27%] Built target RtspServer
make: *** [Makefile:136: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

connortechnology, 2025-01-27:
Good news, your problem is that your system doesn't have the pod2man command.  Normally this just comes with perl, Manjaro must package it separately.  Install it and you should be good to go.

coreybruce, 2025-01-28:
Oh awesome thanks for letting me know, I will install that package from the AUR later on and report how it goes, I'll need to change the pkgbuild file before building it as it only has x86_64 and also ask them to add aarch64 to the package.

coreybruce, 2025-01-29:
> Good news, your problem is that your system doesn't have the pod2man command. Normally this just comes with perl, Manjaro must package it separately. Install it and you should be good to go.

Yep that was exactly what I was missing, I just installed the AUR package and it successfully compiled

I enabled the systemd service but I got this issue

```
× zoneminder.service - ZoneMinder CCTV recording and security system
     Loaded: loaded (/usr/lib/systemd/system/zoneminder.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Wed 2025-01-29 11:53:16 AEST; 3min 49s ago
 Invocation: e5b852354e434a309f5b3976f48fdcf7
    Process: 41361 ExecStart=/usr/bin/zmpkg.pl start (code=exited, status=255/EXCEPTION)
        CPU: 87ms

Jan 29 11:53:16 corey-pi5 zmpkg[41361]: ERR [Error reconnecting to db: errstr:Can't connect to local server through socket '/run/mysqld/mysqld.sock' (2) error val:]
Jan 29 11:53:16 corey-pi5 zmpkg.pl[41361]: Unable to connect to DB. ZM Cannot continue.
Jan 29 11:53:16 corey-pi5 zmpkg.pl[41361]: BEGIN failed--compilation aborted at /usr/share/perl5/vendor_perl/ZoneMinder/Config.pm line 150.
Jan 29 11:53:16 corey-pi5 zmpkg.pl[41361]: Compilation failed in require at /usr/share/perl5/vendor_perl/ZoneMinder.pm line 33.
Jan 29 11:53:16 corey-pi5 zmpkg.pl[41361]: BEGIN failed--compilation aborted at /usr/share/perl5/vendor_perl/ZoneMinder.pm line 33.
Jan 29 11:53:16 corey-pi5 zmpkg.pl[41361]: Compilation failed in require at /usr/bin/zmpkg.pl line 34.
Jan 29 11:53:16 corey-pi5 zmpkg.pl[41361]: BEGIN failed--compilation aborted at /usr/bin/zmpkg.pl line 34.
Jan 29 11:53:16 corey-pi5 systemd[1]: zoneminder.service: Control process exited, code=exited, status=255/EXCEPTION
Jan 29 11:53:16 corey-pi5 systemd[1]: zoneminder.service: Failed with result 'exit-code'.
Jan 29 11:53:16 corey-pi5 systemd[1]: Failed to start ZoneMinder CCTV recording and security system.
```

I assume this is a totally different issue that has nothing to do with this issue tho haha 😆

connortechnology, 2025-01-29:
Looks like you havn't setup the mysql db.

connortechnology, 2025-02-21:
Yup, issue was a missing pod2man.

closed 2025-02-21 by connortechnology