ZoneMinder/zoneminder #4208: Todo: Full processing of the second stream from the camera

IgorA100, 2025-01-27:
I think that full processing of the second stream from the camera is necessary, and not just audio processing.
This would eliminate some problems/inconveniences for high-resolution cameras:

- On Montage Page it is impossible to display RTSP stream simultaneously for 5 or more cameras, because the stream from each camera is 10-15 Mbit/s, and most Internet channels (not counting the local network) have a bandwidth of 30-100 Mbit/s. If ffmpeg is used for display, then a very powerful processor is required so that its core is not loaded to 100%
- There will be no need to create a second monitor for the same camera to analyze motion detection.

I suggest discussing this problem.
I could implement the Frontend part, but I am not very familiar with the Backend part yet.

connortechnology, 2025-01-27:
Mostly just needs Monitor elements to which stream to record and which to do analysis on. I recently added code to select a different video stream when the container contains multiple video streams so adding another couple lines is trivial.

Live view needs to transition away from mjpeg.  Need to switch to webrtc.  But that is a separate issue.

IgorA100, 2025-01-28:
> which stream to record and which to do analysis on

Yes.

> I recently added code to select a different video stream

Great!
Are there any methods I can call via AJAX?

> Need to switch to webrtc.

Yes, of course, I agree.

IgorA100, 2025-02-01:
@connortechnology
If you could make changes to the file "src/zm_monitor_rtsp2web.cpp" I would be grateful!
It is necessary to add a second stream to the "config.json" file belonging to RTSPToWeb
As a result, the "config.json" file should look something like this:

```
    "6": {
      "channels": {
        "0": {
          "name": "ch1",
          "on_demand": true,
          "url": "rtsp://name:pass@1ХХ.1ХХ.1ХХ.2ХХ:554/main"
        },
        "1": {
          "name": "ch2",
          "on_demand": true,
          "url": "rtsp://name:pass@1ХХ.1ХХ.1ХХ.2ХХ:554/third"
        }
      },
      "name": "Camera_Name"
    }
```
rtsp://name:pass@1ХХ.1ХХ.1ХХ.2ХХ:554/main is: "rtsp_path"

Everything else To support displaying the second channel using RTSPToWeb I will do it myself.

IgorA100, 2025-02-01:
To transmit audio, you need to add one line to the parameters.
That is, it should look like this:
```
    "6": {
      "channels": {
        "0": {
          "name": "ch1",
          "audio": true,
          "on_demand": true,
          "url": "rtsp://name:pass@1ХХ.1ХХ.1ХХ.2ХХ:554/main"
        },
        "1": {
          "name": "ch2",
          "audio": true,
          "on_demand": true,
          "url": "rtsp://name:pass@1ХХ.1ХХ.1ХХ.2ХХ:554/third"
        }
      },
      "name": "Camera_Name"
    }
```

IgorA100, 2025-02-12:
API: https://github.com/deepch/RTSPtoWeb/blob/master/docs/api.md

connortechnology, 2025-02-13:
see 50454f26c

IgorA100, 2025-02-13:
> see 50454f26c

Thanks, I saw it.
I left a comment on your code, maybe a small correction should be made.
I created a PR for audio stream support https://github.com/ZoneMinder/zoneminder/pull/4221

I will finish with Rtsp2Web type MSE and make support for the second stream on the front end.

IgorA100, 2025-02-22:
https://github.com/ZoneMinder/zoneminder/pull/4226

IgorA100, 2025-03-04:
I'm closing it because it seems like the second stream is working correctly for RTSP2Web.

closed 2025-03-04 by IgorA100