ZoneMinder/zoneminder #4216: Memory leak after update version zoneminder:amd64

ipelykh, 2025-01-31:
Hello.

After updating the version of zm, installed from ppa.launchpadcontent.net/iconnor/zoneminder-master/ubuntu noble, the memory started leaking.
I am testing ubuntu 24.04 with three 8MP cameras (4k image, Hiseeu HB918-PA, ffmpeg source), 10FPS, 1 sec keyframe with a maximum number of frames in the buffer 20 at each. MoDect, Analysis always. Recording passthrow.
Before the update (with version 1.37.65e20250102141650-noble), all was fine and each zmc process consumed no more than 5 GB of memory, now with version 1.37.65e20250128172246-noble, the memory consumed by zmc processes is constantly growing (10..15..20 GB) quickly after the start zm indefinitely, until all available machine memory runs out, including swap.
I was just doing an apt-get upgrade. Maybe I upgraded incorrectly? At log all is good, no err and war messages. 
I am ready give anything that is needed.

opoplawski, 2025-02-26:
I think we're seeing the zmc process memory grow as well, though perhaps not terribly fast.  At start we see virt size between 444k and 591k.  After running for a while we see some hitting 790k.

zoneminder-1.36.35-1.el9.x86_64

ipelykh, 2025-02-27:
In my case, the zmc process can consume tens of GB, here's the proof:
https://forums.zoneminder.com/viewtopic.php?p=136845#p136845