ZoneMinder/zoneminder #4219: Github master version changed to 1.36.x from 1.37.x?

iamnacho, 2025-02-12:
A few months ago I cloned the master repository and running git describe --tags showed a version like: 1.37.61-654-g2efacc634
(1.37.61 + latest commit)

Today I did the same and master is showing: 1.36.15-4566-gfae724a2c
(1.37.15 + latest commit)

Why the decrease in version number?  This breaks packaging right?

connortechnology, 2025-02-13:
Not sure why it would go back to .15 but I just tagged 1.37.65...

iamnacho, 2025-02-14:
I'm still seeing 1.36.15.
I ran:
git clone https://github.com/ZoneMinder/zoneminder.git
cd zoneminder
git describe --tags

and the output shows
1.36.15-4593-g863ca2109

connortechnology, 2025-02-14:
I don't know, I'm stumped. 
iconnor@pseudo ~/sandbox/ZoneMinder (master)$ git describe --tags
1.37.65-153-g863ca2109

iamnacho, 2025-02-14:
I tried doing a git clone from another machine with the same results.

I also tried:
git clone --branch master --single-branch https://github.com/ZoneMinder/zoneminder.git
& it showed the same 1.36.15

git clone  https://github.com/ZoneMinder/zoneminder.git --branch=1.37.65
reports "fatal: Remote branch 1.37.65 not found in upstream origin"

When I go to:
https://github.com/ZoneMinder/zoneminder

Then Code -> the branch defaults to "master". The I see 32 branches and 93 tags.  When I click on 93 Tags it shows the newest tag is 1.36.35

It's like the 1.37.x tags are completely gone??

rbiegel, 2025-03-18:
I just wondered why 1.37.x is mentioned across the web and I'm not able to locate corresponding downloads or tags. Will we get them back? :face_holding_back_tears:

connortechnology, 2025-03-18:
We don't generally tag dev branch.  We only bump version when there is a db change.  I'm cool with starting to tag when we do that, but so far havn't gotten in the habit. 

There is a lot of effort that goes into a formal release.  So we don't do it.

closed 2026-01-02 by SteveGilvarry