ZoneMinder/zoneminder #4225: "Not running" filter don't work...

onlyjob, 2025-02-21:
ZoneMinder 1.36.35 with PHP8.2 on Debian 12 "bookworm" (Debian package).

I have several "Not Running" (Function=None) monitors but selecting "Not Running" Status in console shows no cameras at all. (Filtering by "Capturing" Status works as expected, showing only the running cameras.)

connortechnology, 2025-02-24:
I agree in terms of intuitiveness.. however ... monitors with Function=None have NO status.  There is no entry in the status column.  (In fact there is no row in the status table), so... this would take a special case.

I'm going to think about this for a bit.

SteveGilvarry, 2026-01-02:
@connortechnology any thoughts?

connortechnology, 2026-01-02:
This should be trivial to fix.  I kinda botched the 1.36.37 release, so doing another .38 is kinda needed anyways.